mavlink/mavros #837: Sending waypoint info by using mavros services

aykutkabaoglu, 2017-10-15:
Hi,
I am trying to use mavros/services to send some waypoint information to PX4. The pixhawk is connected via USB and mavros communication is set. I can send waypoint data by command line as 
```
rosservice call /mavros/mission/push "start_index: 0
waypoints:
- {frame: 0, command: 16, is_current: false, autocontinue: false, param1: 0.0, param2: 0.0,
  param3: 0.0, param4: 0.0, x_lat: 10.0, y_long: 20.0, z_alt: 30.0}" 
```

However, whenever i call the services from a cpp code, i failed. I put three different services in a node to test where the fault is.

```
#include <ros/ros.h>
#include <mavros_msgs/WaypointPush.h>
#include <mavros_msgs/WaypointClear.h>
#include <mavros_msgs/CommandHome.h>
#include <std_msgs/String.h>
#include <cstdlib>
#include <mavros_msgs/Waypoint.h>

int main(int argc, char **argv)
{
  ros::init(argc, argv, "srv_waypoint");
  ros::NodeHandle p;
  ros::NodeHandle n;
  ros::NodeHandle l;

  ros::ServiceClient wp_clear_client = p.serviceClient<mavros_msgs::WaypointClear>("waypoint_clear_client");
  ros::ServiceClient wp_srv_client = n.serviceClient<mavros_msgs::WaypointPush>("waypoint_push_client");
  ros::ServiceClient set_home_client = l.serviceClient<mavros_msgs::CommandHome>("command_home_client");

  mavros_msgs::WaypointPush wp_push_srv;
  mavros_msgs::WaypointClear wp_clear_srv;
  mavros_msgs::CommandHome set_home_srv;

  mavros_msgs::Waypoint wp_msg;

  wp_msg.frame = 0; // mavros_msgs::Waypoint::FRAME_GLOBAL;
  wp_msg.command = 16;
  wp_msg.is_current = false;
  wp_msg.autocontinue = false;
  wp_msg.param1 = 0;
  wp_msg.param2 = 0;
  wp_msg.param3 = 0;
  wp_msg.param4 = 0;
  wp_msg.x_lat = 40.0;
  wp_msg.y_long = 30.0;
  wp_msg.z_alt = 5.0;

  wp_clear_srv.request = {};

  wp_push_srv.request.start_index = 0;
  wp_push_srv.request.waypoints.push_back(wp_msg);

  set_home_srv.request.current_gps = false;
  set_home_srv.request.latitude = 11;
  set_home_srv.request.longitude = 12;
  set_home_srv.request.altitude = 1;

  if (set_home_client.call(set_home_srv))
{
    ROS_INFO("Home was set to new value ");
}
else
{
    ROS_ERROR("Home position couldn't been changed");
}

if (wp_clear_client.call(wp_clear_srv))
{
    ROS_INFO("Waypoint list was cleared");
}
else
{
    ROS_ERROR("Waypoint list couldn't been cleared");
}

  if (wp_srv_client.call(wp_push_srv))
  {
    ROS_INFO("Success:%d", (bool)wp_push_srv.response.success);
  }
  else
  {
    ROS_ERROR("Waypoint couldn't been sent");
    ROS_INFO("Success:%d", (bool)wp_push_srv.response.success);
  }

   ros::spinOnce();

  return 0;
}
```

And the program output is below, none of service calls were done. I investigated all mavros documentation about services but i couldn't figure out where the problem is.

```
[ERROR] [1508100486.320659255]: Home position couldn't been changed
[ERROR] [1508100486.321302729]: Waypoint list couldn't been cleared
[ERROR] [1508100486.321785823]: Waypoint couldn't been sent
[ INFO] [1508100486.321812804]: Success:0
```

Thanks,

### MAVROS version and platform

Mavros: 0.21.1
ROS: Kinetic
Ubuntu: 16.04

vooon, 2017-10-15:
`NodeHandle::serviceClient()` take service path in first arg.

aykutkabaoglu, 2017-10-16:
Thank you, i figured it out now. 
To explain it clearly for others, i changed `("waypoint_clear_client")` as `("/mavros/mission/clear")`

` ros::ServiceClient wp_clear_client = p.serviceClient<mavros_msgs::WaypointClear>("waypoint_clear_client");`

` ros::ServiceClient wp_clear_client = p.serviceClient<mavros_msgs::WaypointClear>("/mavros/mission/clear");`

closed 2017-10-17 by vooon

debanik123, 2020-06-22:
#!/usr/bin/env python

import rospy
from std_msgs.msg import String
from sensor_msgs.msg import NavSatFix
from mavros_msgs.msg import *
from mavros_msgs.srv import *

def create_waypoint():
    
	waypoint_clear_client()
	wl = []
	wp = Waypoint()

	wp.frame = 3
	wp.command = 22  # takeoff
	wp.is_current = False
	wp.autocontinue = True
	wp.param1 = 0  # takeoff altitude
	wp.param2 = 0
	wp.param3 = 0
	wp.param4 = 0
	wp.x_lat = -35.363578
	wp.y_long = 149.1656228
	wp.z_alt = 0
	wl.append(wp)
    
	
	wp = Waypoint() 

	wp.frame = 3
	wp.command = 16  #Navigate to waypoint.
	wp.is_current = False
	wp.autocontinue = True
	wp.param1 = 0  # delay 
	wp.param2 = 0
	wp.param3 = 0
	wp.param4 = 0
	wp.x_lat = -35.363582
	wp.y_long = 149.1656232
	wp.z_alt = 0
	wl.append(wp)

    
	wp = Waypoint()

	wp.frame = 3
	wp.command = 19  # Navigate to waypoint.
	wp.is_current = False
	wp.autocontinue = True
	wp.param1 = 0  # delay
	wp.param2 = 0
	wp.param3 = 0
	wp.param4 = 0
	wp.x_lat = -35.363586
	wp.y_long = 149.1656236
	wp.z_alt = 0
	wl.append(wp)
	
	

	wp = Waypoint()

	wp.frame = 3
	wp.command = 19  # takeoff
	wp.is_current = False
	wp.autocontinue = True
	wp.param1 = 0  # takeoff altitude
	wp.param2 = 0
	wp.param3 = 0
	wp.param4 = 0
	wp.x_lat = -35.20
	wp.y_long = 149.1656240
	wp.z_alt = 0
	wl.append(wp)

	wp = Waypoint()

	wp.frame = 3
	wp.command = 19  # takeoff
	wp.is_current = False
	wp.autocontinue = True
	wp.param1 = 0  # takeoff altitude
	wp.param2 = 0
	wp.param3 = 0
	wp.param4 = 0
	wp.x_lat = -35.40
	wp.y_long = 149.1656244
	wp.z_alt = 0
	wl.append(wp)

	wp = Waypoint()

	wp.frame = 3
	wp.command = 19  # takeoff
	wp.is_current = False
	wp.autocontinue = True
	wp.param1 = 0 # takeoff altitude
	wp.param2 = 0
	wp.param3 = 0
	wp.param4 = 0
	wp.x_lat = -35.50
	wp.y_long = 149.1656248
	wp.z_alt = 0
	wl.append(wp)

	print(wl)
	
	try:
	    service = rospy.ServiceProxy('mavros/mission/push', WaypointPush, persistent=True)
	    service(start_index=0, waypoints=wl)
	    

	  
	    if service.call(wl).success: #burasi belki list istiyordur. birde oyle dene
	        print 'write mission success'
	    else:
	        print 'write mission error'
	  
	except rospy.ServiceException, e:
	    print "Service call failed: %s" % e

	

def waypoint_clear_client():
        try:
            response = rospy.ServiceProxy('mavros/mission/clear', WaypointClear)
            return response.call().success
        except rospy.ServiceException, e:
            print "Service call failed: %s" % e
            return False

if __name__ == '__main__':
	rospy.init_node('waypoint_node', anonymous=True)
	pub = rospy.Publisher('global',String,queue_size=10)
	konum = "suan buradasin" # to be used later
	pub.publish(konum)
	create_waypoint()

debanik123, 2020-06-22:
waypoint_mission from python node

[waypoint_mission_ok.zip](https://github.com/mavlink/mavros/files/4815184/waypoint_mission_ok.zip)

debanik123, 2020-06-22:
Download this zip file and go to go.