mavlink/mavros #838: Timesync issue when using Snapdragon Flight with mavros

bkatona, 2017-10-18:
We are using a Snapdragon Flight to pass vision-based pose information to PX4 with mavros over a serial connection at 921600 baud. When we run the VISLAM ROS node and publish to /mavros/vision_pose/pose, we get the [lpe] vision position timeout error continually. We have the timesync_mode set to PASSTHROUGH based on the discussion in https://github.com/PX4/Firmware/issues/6562. The other time timesync modes have the same result. Upon inspection the ROS topic timestamp and PX4 timestamp still seem to differ by large orders of magnitude. How can the difference between these timestamps be handled?

### Autopilot type and version
Mavros: 0.21.2
ROS: Indigo
Ubuntu: 14.04 (Linaro)
PX4: 1.6.5

### Node logs
```SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_url: /dev/ttyUSB0:921600
 * /mavros/gcs_url: udp://@192.168.1....
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
 * /mavros/global_position/tf/frame_id: map
 * /mavros/global_position/tf/global_frame_id: earth
 * /mavros/global_position/tf/send: False
 * /mavros/global_position/use_relative_alt: True
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
[... 348 lines cut ...]
    hardware_id: /dev/ttyUSB0:921600
    values: 
      - 
        key: Voltage
        value: 15.01
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 100.0

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 5388 messages, from 1 addresses
sys:comp   list of messages
  1:1     32, 1, 2, 36, 230, 102, 105, 74, 140, 141, 77, 111, 76, 241, 147, 245, 22, 0, 253, 30, 31

```

TSC21, 2017-10-18:
At what rate are you publishing the vision estimate? The minimal rate required for vision on LPE is 2hz.

bkatona, 2017-10-18:
The vision estimate is being published at 30 hz.

TSC21, 2017-10-20:
Check your connection to the FCU then. If the problem is not on the ROS side, it has to be on the link. Try to increase the timeout [here](https://github.com/PX4/Firmware/blob/master/src/modules/local_position_estimator/sensors/vision.cpp#L15).

bkatona, 2017-10-20:
We tried increasing the timeout to 1 second as well as 10 seconds. Both still result in the timeout error. It seems that changing the timeout value will not matter based on the timestamps we are seeing. If we print out (below) the timestamps used in the timeout check [here](https://github.com/PX4/Firmware/blob/c9f44531c20c81500f59680a3e1bc06963900976/src/modules/local_position_estimator/sensors/vision.cpp#L145) we can see that the difference is much larger than 1 second or even 10 seconds (delta is _timeStamp - _time_last_vision_p)

We should note too that running rostopic hz /mavros/vision_pose/pose reports 30 hz. Looking at the vision_position_estimate message in the QGroundControl Mavlink Inspector, we see that PX4 is also receiving the messages at 30 hz.

```
[ERROR] [1508530012.854234002]: FCU: [lpe] vision position timeout
[ERROR] [1508530012.857109698]: FCU: [lpe] _timeStamp 850125517
[ERROR] [1508530012.859273201]: FCU: [lpe] _time_last 18445235544793063424
[ERROR] [1508530012.865921838]: FCU: [lpe] delta 1508529766613423
[ INFO] [1508530012.871907815]: FCU: [lpe] vision position init: -0.00 0.00 0.00 m s
[ERROR] [1508530012.881256104]: FCU: [lpe] vision position timeout
[ERROR] [1508530012.885233422]: FCU: [lpe] _timeStamp 850175513
[ERROR] [1508530012.891472681]: FCU: [lpe] _time_last 18445235544793131008
[ERROR] [1508530012.901296755]: FCU: [lpe] delta 1508529766597080
[ INFO] [1508530012.932335392]: FCU: [lpe] vision position init: -0.00 0.00 -0.00 m s
[ERROR] [1508530012.943341661]: FCU: [lpe] vision position timeout
[ERROR] [1508530012.946648557]: FCU: [lpe] _timeStamp 850234231
[ERROR] [1508530012.950442958]: FCU: [lpe] _time_last 18445235544793196544
[ERROR] [1508530012.957531076]: FCU: [lpe] delta 1508529766588586
[ INFO] [1508530012.996294347]: FCU: [lpe] vision position init: -0.00 0.00 -0.00 m s
[ERROR] [1508530013.003350018]: FCU: [lpe] vision position timeout
[ERROR] [1508530013.007936610]: FCU: [lpe] _timeStamp 850298381
[ERROR] [1508530013.012423931]: FCU: [lpe] _time_last 18445235544793264128
[ERROR] [1508530013.015855307]: FCU: [lpe] delta 1508529766586380
```

TSC21, 2017-10-21:
Yeah there seems to be a big difference between timestamps. How are you publishing your timestamps on the vision msg? What's the format?

bkatona, 2017-10-23:
The timestamps are being published in seconds as a ros::Time type on the vision message [here](https://github.com/bkatona/ros-examples/blob/6850597eee0793af3f3239ca5592108a13e028a0/src/nodes/SnapdragonRosNodeVislam.cpp#L204-L211).

TSC21, 2017-11-05:
@bkatona how's this going to you atm? Are you using NTP for timesync?

bkatona, 2017-11-07:
No change in recent weeks. 
We do have chrony running on the Snapdragon Flight, but we have not done anything specific to use it to sync with the Pixhawk. We were relying on the mavros timesync function to perform the timesync. Is there a way to do it with NTP?

TSC21, 2018-01-28:
@bkatona updates?

bkatona, 2018-01-29:
We have moved away from using the LPE on PX4, as updates to EKF2 have allowed us to accomplish what we need.

TSC21, 2018-01-29:
I was refering to the timesync itself.

closed 2018-01-29 by TSC21

bkatona, 2018-01-29:
Last time we tried, we still saw the same error.

gssammy, 2019-05-20:
@bkatona Can you point me to any documentation you used to set up vislam on the ekf2 and px4? I suspect i'm still seeing some slight delays in the vislam pose that the estimator is not correctly compensating for.