mavlink/mavros #840: Is there a keyboard control using Mavros for PX4 Drone without GPS?

vinhk, 2017-10-19:
Hi,

Is there a keyboard code using MAVROS that can be tested on the PX4 quadcopter without GPS or other external position. I am aware that without position it will be difficult to calculate errors. For now, that is fine. I just wanted to be able to make the quadcopter respond to the commands. 

Thank you

TSC21, 2017-10-20:
@vinhk I have a WIP mode for joystick script that will allow keyboard control over the vehicle, but I left it stale as I have other priorities. For now, if you want to do manual control over mavros, you can use a gamepad. Check an example of configuration [here](https://github.com/mavlink/mavros/blob/master/mavros_extras/launch/f710_joy.yaml). An example launch file can be found [here](https://github.com/mavlink/mavros/blob/master/mavros_extras/launch/teleop.launch).

vinhk, 2017-10-20:
is 'joy_node' a standard ros joy node from [here](https://github.com/ros-drivers/joystick_drivers)?

TSC21, 2017-10-20:
I guess, yes.

vinhk, 2017-10-20:
Thanks, I will test it soon :)

closed 2017-10-20 by TSC21

vinhk, 2017-12-05:
@TSC21 is there a way to arm it. I can connect but nothing happens yet. Do you know how to arm px4 drone using the gamepad f710?