mavlink/mavros #841: Why we need to send every 5 hz?

vinhk, 2017-10-20:
I have this question regarding sending setpoint every 5 hz or otherwise the FCU will reject the command or code we are running using MAVROS. Is this not the same thing as this: 

```c++
ros::Rate rate(5);
while(ros::ok())
{
    rate.sleep();
}
```

or it should be implemented in a launch file like this: 
```xml
<node pkg="joy" type="joy_node" name="joy" required="True">
		<param name="autorepeat_rate" value="5" /> <!-- Minimal update rate, Hz -->
	</node>
```
if someone can give more info on this topic would be tremendous.
Thanks

TSC21, 2017-10-20:
> I have this question regarding sending setpoint every 5 hz or otherwise the FCU will reject the command or code we are running using MAVROS.

where do you see that?

vinhk, 2017-10-20:
I can not remember exactly. But there were someplace I read where you must update setpoint every 5 seconds or so. Correct me if I am wrong.

vinhk, 2017-10-20:
I think it was this [one ](https://404warehouse.net/2015/12/20/autopilot-offboard-control-using-mavros-package-on-ros/), and a few others places on Github.

> Setpoints should be sent to the flight controller with a 0.5s time out. This is a safety factor so that the control should converge to a specific objective. Otherwise the flight controller will not go into offboard mode.

TSC21, 2017-10-20:
@vinhk 0.5s = 2hz. 5hz equals to 0.2s. And it is set to 2Hz (0.5s) as a requirement on the Firmware: https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L150.

closed 2017-10-20 by TSC21