mavlink/mavros #845: ypv

ypvypv, 2017-10-29:
Hi guys,

I am new to mavros but I am thrilled to harness its power to control drone's position, state, heading autonomously.

I followed the offboard tutorial to control gazebo px4 from mavros node to set waypoints and that is working well. However what I thought would be a reasonable workflow to add heading/attitude to the drone while following waypoints turned out to be a difficult exercise.

I publish/subscribe to /mavros/setpoint_attitude/target_attitude. I am also using latest source mavros. I configured the yaml to use_quaternion=true and I am seeing callbacks firing when quaternion attitude is passed in.

I hardcode a reasonable 90 degree around-z quaternion :
  orientation: 
    x: 0.0
    y: 0.0
    z: -0.707106769085
    w: 0.707106769085

When I combine waypoints to mavros/setpoint_position/local and attitude to /mavros/setpoint_attitude/target_attitude

I get random drone turns and as it becomes unstable it crash-lands. My feeling is this is product of incorrect fusing of set position and set attitude, resulting in incoherent /mavros/imu/data or /mavros/local_position/pose.

When I read /mavros/local_position/pose I see that the orientation goes back and forth between stable constant orientation, what I perceive as result of mavros/setpoint_position/local, and /mavros/setpoint_attitude/target_attitude.

Please let me know if I am expecting too much from attitude handling of mavros but here is what I expect: I want to set target attitude to point in a given quaternion direction ( with identity quaternion being constructed from vector3( 1.0f, 0.0f, 0.0f )   ) and the drone should orient itself toward east. Getting frame of reference right is secondary. I would just like to say, make a 90 degree or 180 degree turn without losing altitude. Ideally, I want to reference a quaternion representing a rotation from 0 rotation and have drone follow that rotation. 

Many thanks in advance for any assistance!

Thanks,
ypv

### MAVROS version and platform

Mavros: 0.21.2
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ * ] PX4

Version: v1.6.5 ( I believe )

### Node logs

```
roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557" gcs_url:="udp://@172.17.0.1"
... logging to /home/user/.ros/log/3e575b9c-bcdb-11e7-9c18-00dbdf670b9c/roslaunch-ypv-Inspiron-7559-6064.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ypv-Inspiron-7559:38517/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
[... 411 lines cut ...]
      - 
        key: Mean dt (ms)
        value: -1.983774
      - 
        key: Last system time (s)
        value: 174.954356000
      - 
        key: Time offset (s)
        value: 1509303486.993827581
---

```

### Check ID

```
user@ypv-Inspiron-7559:/$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 5823 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 140, 141, 147, 24, 30, 31, 32, 33, 74, 76, 85, 87, 102, 230, 105, 106, 111, 241, 242, 245

```

ypvypv, 2017-11-04:
Can anyone please comment at least in the general idea of the solution, applying orientation in offboard mode? Is there a better method?

TSC21, 2017-11-05:
@ypvypv you cannot use `setpoint_altitude` at the same time as `setpoint_position`. You use one or another. If you want to control yaw, you can control it by value or by rate. With `setpoint_position` you can control `position`+`yaw`(heading). With `setpoint_velocity` you can control `velocity`+`yaw_rate`. With `setpoint_raw`, depending on the bitmask you use on your msg, you can do multiple types of control (position+yaw_rate, velocity+yaw, XY_velocity+Z_height, etc.). But for attitude control, you cannot use just the orientation quaternions, you also need to publish the respective thrust value.

closed 2017-11-08 by TSC21

ypvypv, 2017-11-16:
Just to add more information for the solution I found. It turned out that inside /mavros/setpoint_position/local, position and orientation were everything that I needed. Once I fixed my quaternion interpolation, orientation worked very well.

mandelyoni, 2018-06-23:
Dear Sir,

We're running in Offboard mode and sending a set_attitude_target message through Mavlink.
The PX4 version is 1.6.5 on Intel Aero.

It seems that the roll and pitch angle/rate commands work fine but the yaw angle/rate doesn't work at all. 

Please advise