mavlink/mavros #846: Kinetic+PX4+Telem2USB+OrdoidXU4 cannot launch (resource deadlock)

fbalaban, 2017-10-30:
### Issue details

Cannot launch node(kinetic) running on odroidXU4 and connected to pixhawk at telemetry2 port. Mavproxy runs correctly, connected at 921600. Since new version of ardupilot(arduplane) does not have a SYS_COMPANION parameter, the baud rate has been set up in parameter SERIAL2_BAUD (value 921). The protocol parameter is SERIAL2_PROTOCOL 1

### MAVROS version and platform

Mavros: 0.21.2
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: 3.8.2

### Node logs

```
odroid@odroid:~/Experiments/Temp$ roslaunch mavros px4.launch
... logging to /home/odroid/.ros/log/ba162802-d0e3-11e5-8d7b-9731d3d54c31/roslaunch-odroid-5369.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://odroid:53720/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/component_id: 1
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
[... 154 lines cut ...]
header:
    seq: 0
    stamp:
        secs:  1509386011
        nsecs:  138879442
    frame_id: ' '
status:
    -
        level:  0
        name:  mavros:  FCU connection
        message:  Node starting up
        hardware_id: ' '
        values:  []
---
```

### Check ID

```
ERROR. I got 0 addresses, but not on your target 1:1

---
Received 0 messages, from 0 addresses
sys:comp        list of messages
```

vooon, 2017-10-30:
Your problem in locked uart, try to reboot.
Mentioned parameters is from APM, you sure that you flash right firmware?

fbalaban, 2017-10-31:
Sorry, I have two pixhawks one with apm and other with px4 and mixed the logs. Now I've edited my original post. Same thing happens. The firmware uploaded is apm. I will also try with px4 and let you know. I've rebooted the autopilot btw.

khancyr, 2017-10-31:
You should check your wiring (TX  - RX - GND) and check that your powering is stable

fbalaban, 2017-10-31:
Nope, same thing happens (similar mavros side logs) no matter if the firmware is apm or px4. Since I'm having communication through mavproxy, shouldn't it work? I don't know if it matters, but the autopilot is powered by the usb port to another computer (not the one trying to act like a companion). Moreover, the TELEM1 has the usual telemetry radio plugged in.

TSC21, 2017-10-31:
> I don't know if it matters, but the autopilot is powered by the usb port to another computer (not the one trying to act like a companion)

This may interfere with the logic levels of the serial interface.

khancyr, 2017-10-31:
You need common ground linking ! 
But you can use the serial port with both mavproxy and mavros at the same time

fbalaban, 2017-10-31:
I'm sorry but I'm not much of an electronics guy. Does this mean that I'll have to power pixhawk and odroid from the same source, similar to figure "Odroid Power Connections" in [that page](http://ardupilot.org/dev/docs/odroid-via-mavlink.html)? I thought that since mavproxy was working fine, I wouldn't have a problem with the telem2 communication.

fbalaban, 2017-10-31:
Same behaviour powered by a battery and only having connected the telem2 port.

fbalaban, 2017-11-01:
So, after interchanging everything (cables, FCU's, firmware, ports etc) I've tested the mavros connection only by using the usb port of pixhawk and the same happened. It managed to work only by rebooting everything without any mavproxy, qgroundcontrol, or apmPlanner running before it. Probably something locks the serial port, even though it doesn't seem to stop the other applications. It seems that the issue is the same as [issue 642](https://github.com/mavlink/mavros/issues/642). 

I'm sorry @vooon , when you've said restart, I didn't realise that you meant the companion, I thought you meant the autopilot, after the baudrate change. I guess you can close the issue, but if there is something else to be done I would be happy to know it.

closed 2017-11-01 by vooon