mavlink/mavros #848: Geographiclib target directory not found

kandai-wata, 2017-11-05:
Hi, I'm new to pixhawk and mavros. 
I followed the [instruction](https://github.com/mavlink/mavros/blob/master/mavros/README.md#required-dependencies) to install mavros, but I got an error as below.

### Issue details
when I ran `install_geographiclib_datasets.sh` or `geographiclib-datasets-download egm96_5`
I get 
`Geographiclib target directory /usr/share/geographiclib not found` 

During the process, I did few things.
1. I once downloaded from binary source and apt removed it 
2. Although my ROS version is Indigo, I added `--rosdistro kinetic` when I ran `rosinstall_generator`  as commented in this [issue](https://github.com/mavlink/mavros/issues/600)
How should I install the geographic data?

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Indigo?
Ubuntu: ?14.04?

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

TSC21, 2017-11-05:
`sudo apt-get install libgeographic-dev`
`sudo apt-get install geographiclib-tools`

TSC21, 2017-11-05:
A note:
> I'm using 3DR Solo with the initial setup. So I guess it's PX4 based (am I right?) with APM on it.

3DR Solo flight controller is Pixhawk 2.1 (Cube). PX4 and APM are fligh stacks that may run on it.

kandai-wata, 2017-11-05:
> sudo apt-get install libgeographic-dev
> sudo apt-get install geographiclib-tools
 
Thank you for your help!
I tried it and successfully installed geographiclib-tools, 
but it says it cannot find libgeographic-dev. 

> 3DR Solo flight controller is Pixhawk 2.1 (Cube). PX4 and APM are fligh stacks that may run on it.

I am assuming that Hardware is Pixhawk 2.1 and the software is PX4 or APM.
However, for software, I have [read](https://github.com/3drobotics/ardupilot-solo/issues/210) that to upload a firmware,
I need to git clone
```
https://github.com/3drobotics/ardupilot-solo
https://github.com/3drobotics/PX4Firmware-solo
https://github.com/3drobotics/PX4NuttX
```
and eventually `make` all of these.

TSC21, 2017-11-05:
Sorry. Saw now you are with 14.04. Then issue `sudo apt-get install libgeographiclib-dev`.

kandai-wata, 2017-11-05:
Thank you TSC21.
`sudo apt-get install libgeographiclib-dev` did work.
However, it still cannot install geographic datasets.

```
watanabe@ytnpc2016a-W230SD:~/catkin_ws$ sudo ./src/mavros/mavros/scripts/install_geographiclib_datasets.sh
[sudo] password for watanabe: 
+ [[ != 0 ]]
./src/mavros/mavros/scripts/install_geographiclib_datasets.sh: 7: ./src/mavros/mavros/scripts/install_geographiclib_datasets.sh: [[: not found
+ which /usr/sbin/geographiclib-get-geoids
+ which geographiclib-datasets-download
/usr/sbin/geographiclib-datasets-download
+ geographiclib-datasets-download egm96_5
Geographiclib target directory /usr/share/geographiclib not found

```

TSC21, 2017-11-05:
I can't really tell. Considering you already install GeographicLib and the tools for downloading the datasets, it should work with `geographiclib-datasets-download egm96_5`. I can't test it myself as I'm using Ubuntu 16.04, but the above scripts should have installed the required libs and files. What's the output of `locate geographiclib` or `locate GeographicLib`?

vooon, 2017-11-06:
A bit off: Solo uses Pixhawk 2, not 2.1 (that is made by hex, cube+carrier). And as i know, 3DR cube a bit different from cube made by hex, but mostly compatible.
By default Solo use APM firmware (also special version, but i think that vanilla should work too).

Try just create that directory `sudo mkdir -p /usr/share/geographiclib`, also check where `libgeographic.so` is located.

kandai-wata, 2017-11-06:
I tried `sudo mkdir -p /usr/share/geographiclib`  and `./src/mavros/mavros/scripts/install_geographiclib_datasets.sh`
I did start to download the geographiclib datasets. 

Then I followed the further instructions written in README.md and tried to roslaunch.
It said it could not find anything in `/usr/share/geographiclib/GeographicLib/`, 
so I `sudo mkdir GeographicLib` and `mv geoids/ GeographicLib/`, and it worked.

By the way,
I could successfully connect to 3DR Solo with
`roslaunch mavros apm.launch fcu_url:=udp://:14550@10.1.1.10`
for those who want to develop with 3DR Solo!

Thank you so much for your help @TSC21 and @vooon

closed 2017-11-06 by kandai-wata