mavlink/mavros #849: uas: libatomic do not exist on MacOS

vooon, 2017-11-07:
As reported by @eric1221bday, mac do not have libatomic, so fix from #797 breaks compilation.
Workaround: comment-out `atomic` from `target_link_libraries(mavros, ...)`.

TSC21, 2017-11-07:
We should find a way to generalize this workaround. The best (but overkill way) would be to add a `.cmake` module to search (or set) the `TARGET` architecture.

eric1221bday, 2017-11-07:
I don’t know how fine grained you would want the target to be, but cmake does provide flags for OS, compiler, and CPU, check out https://cmake.org/cmake/help/v3.6/manual/cmake-toolchains.7.html

vooon, 2017-11-07:
Problem that we also depends on OS version. For example Hydro has similar problem with older GCC.
I also suppose, that newer versions of OSes for ARM may nor require explicit linking with libatomic.
But for now it is needed at least for ARMv7 and ARMv8. AMD64 do not require, and perhaps LLVM don't need explicit libatomic.

eric1221bday, 2017-11-07:
OS version I believe is also part of CMake, there's also a flag for compiler version, https://cmake.org/cmake/help/v3.0/variable/CMAKE_LANG_COMPILER_VERSION.html, however as the page noted it's not guaranteed to defined for every version or compiler, so that may warrant some testing

closed 2021-03-03 by vooon