mavlink/mavros #852: Ubuntu 17.04 not supported

fnoop, 2017-11-08:
### Issue details

Error when trying to build from source on ubuntu 17.04 (zesty)

### MAVROS version and platform

Mavros: master
ROS: Kinetic
Ubuntu: 17.04

### Diagnostics

```
rosdep install --from-paths src --ignore-src -y
dpkg-query: no packages found matching libgeographic-dev
dpkg-query: no packages found matching geographiclib-tools
dpkg-query: no packages found matching python-future
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
mavros: No definition of [geographic_msgs] for OS version [zesty]
mavros_extras: No definition of [tf] for OS version [zesty]
test_mavros: No definition of [control_toolbox] for OS version [zesty]
mavros_msgs: No definition of [geographic_msgs] for OS version [zesty]
```

TSC21, 2017-11-08:
@fnoop why are you trying to build from source with Kinetic on Zesty? Kinetic is not supported as is on Zesty, so you should upgrade to Lunar.

fnoop, 2017-11-08:
Maverick (https://github.com/fnoop/maverick) tries to keep consistent software versions across all platforms where possible, and it's on Kinetic at the moment.  ROS kinetic builds and runs fine from source on unsupported platforms like ubuntu 17.04 and raspberry, it's just mavros I'm having problem with.

TSC21, 2017-11-08:
On rosdep:
https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L862
https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L866
https://github.com/ros/rosdistro/blob/master/rosdep/python.yaml#L1082
On deb package list:
https://packages.ubuntu.com/zesty/libgeographic-dev
https://packages.ubuntu.com/zesty/geographiclib-tools
https://packages.ubuntu.com/zesty/python-future
About the rosdep dependencies: the ROS release is usually compatible with certain OS releases. So if Kinetic only has support of deb packages till Xenial, every other packages it tries to get from the deb packages will have to be defined for that same OS release. You can see that `geographic_msgs`, `tf`, `control_toolbox` and `geographic_msgs` are both supported on Kinetic and Lunar, but Kinetic only goes till 16.04 support. From there, you will have to get everything from source, unless you upgrade the ROS version. That means you have to install `geographic_msgs`, `tf`, `control_toolbox` and `geographic_msgs` also from source. The dependencies above can be installed by `apt-get`.

TSC21, 2017-11-08:
The consistency I advise you to have is keep compatibility between the OS, platforms and applications/frameworks. Not always is a good idea to keep the same SW version across all platforms.

fnoop, 2017-11-08:
OK so following the mavros source install instructions, these miss out the source dependencies:
https://github.com/mavlink/mavros/blob/master/mavros/README.md#installation
I think these should be updated for a proper source install.

TSC21, 2017-11-08:
@fnoop we try to generalize the support as much as possible, even for unsupported platforms. But the install instructions are made for a more common approach of the user/developer. Source install Kinetic on Zesty is not a common procedure. So at a max, we can provide some tips, but we cannot have everything on the Readme file. The dependencies on the common approach install with the `rosdep` command and there is no need for source install it at all. The only thing that the user has to do manually is install the geo datasets, which cannot be done automatically. The rest is handle by `rosdep`.

fnoop, 2017-11-08:
But those instructions miss source dependencies for any ros distro on any platform don't they?

TSC21, 2017-11-08:
Have you tried the following command? `rosdep install --from-paths ${ROS_WORKSPACE} --ignore-src --rosdistro=kinetic`

TSC21, 2017-11-08:
> But those instructions miss source dependencies for any ros distro on any platform don't they?

No they don't. Read what's stated there:
> Most of the ROS dependencies are supported and installed by rosdep, including external libraries as Eigen and Boost.
(...)
Note that if you are using an older MAVROS release source install and want to update to a new one, remember to run `rosdep` update before running `rosdep install --from-paths ${ROS_WORKSPACE} --ignore-src --rosdistro=${ROSDISTRO}`, with `ROS_WORKSPACE` your `src` folder of catkin workspace. This will allow updating the `rosdep` list and install the required dependencies when issuing rosdep install.

TSC21, 2017-11-08:
Basically if you find you cannot install the dependencies with `rosdep`, you will have to install the manually. But that only means you are working with an unsupported platform, where `rosdep` cannot actuate by itself.

fnoop, 2017-11-08:
Won't 'rosinstall_generator --upstream mavros --rosdistro kinetic --deps' generate the correct dependencies?
I'm expecting I'll get the same problems on the raspberry, or any other platform that doesn't have the binary dependencies available.  My mavros puppet recipe used to work fine on unsupported platforms.

TSC21, 2017-11-08:
> I'm expecting I'll get the same problems on the raspberry, or any other platform that doesn't have the binary dependencies available. My mavros puppet recipe used to work fine on unsupported platforms.

Again, you will have to installed them by src. It generates the dependencies for the supported OS where Kinetic works. It's not able to generate for 17.04 for example. There are other people that were able to get MAVROS working on Raspberry Pi and other platforms, even installing ROS from source, but they had to also install the respective dependencies from source. Again, we cannot tackle every platform or situation on the Wiki. Those situations should be part of the research of those who are trying to get something more "exotic" to work. Not everything works out of the box, though there's several sources explaining how to install ROS from source and the required dependencies for each package.

fnoop, 2017-11-08:
These are not exotic requirements, they're basic source installs.  Adding '--deps' to the generator seems to fill in the correct source requirements, as per:
https://github.com/fnoop/maverick/blob/master/manifests/maverick-modules/maverick_ros/manifests/init.pp#L224
Raspberry source build now works, running 17.04 to see if that works as well.

closed 2017-11-08 by TSC21

fnoop, 2017-11-08:
This doesn't fix 17.04, I get merge conflict errors. But you're right, I should just use lunar for 17.04.  Thanks for updating docs - it does fix raspberry build.

TSC21, 2017-11-08:
NP. Yeah I guess this will always represent a problem when the ROS version is not compatible with the OS version. Lunar will for sure work.