mavlink/mavros #853: install_geographiclib_datasets.sh wrong interpreter

fnoop, 2017-11-08:
### Issue details

install_geographiclib_dataset.sh uses /bin/sh as the specified interpreter, but uses bash specific syntax ([[).  Should specify /bin/bash.

### MAVROS version and platform

Mavros: 0.21.3
ROS: Kinetic
Ubuntu: 16.04

```
[dev] [mav@maverick-ubuntuvm /var/tmp]$ ./install_geographiclib_datasets.sh
+ [[ != 0 ]]
./install_geographiclib_datasets.sh: 7: ./install_geographiclib_datasets.sh: [[: not found
+ which /usr/sbin/geographiclib-get-geoids
/usr/sbin/geographiclib-get-geoids
+ run_get geoids geoids egm96-5
+ local dir=geoids
+ local tool=geoids
+ local model=egm96-5
+ [[ -d /usr/share/GeographicLib/geoids
./install_geographiclib_datasets.sh: 18: ./install_geographiclib_datasets.sh: [[: not found
+ -d /usr/local/share/GeographicLib/geoids ]]
./install_geographiclib_datasets.sh: 18: ./install_geographiclib_datasets.sh: -d: not found
```
etc

TSC21, 2017-11-08:
https://github.com/mavlink/mavros/pull/842
I can't say if this introduced a problem, but I am not able to reproduce the problem though.

fnoop, 2017-11-08:
I don't think #842 has anything to do with it.  What does `ls -l /bin/sh` say on your system?

TSC21, 2017-11-08:
Nevermind. I could reproduce it. #854 fixes it.

PR #854 (2017-11-08): geolib_dataset: script: fix interpreter
Fixes #853.
closed 2017-11-08 by TSC21