mavlink/mavros #856: serial0: receive: End of file

pavloblindnology, 2017-11-09:
----

### Issue details

Hi.
Sometimes I get "serial0: receive: End of file" message when running mavros (nothing other connected to autopilot), though I can connect normally through mavproxy, APM Planner and QGroundControl.
Restart of autopilot doesn't help, only restart of PC.

### MAVROS version and platform

Mavros: 0.21.3
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[ + ] ArduPilot
[  ] PX4

Version: APMRover v3.1.2, v.3.2.0-rc1, etc.

### Result of miniterm.py

```
--- Available ports:
---  1: /dev/ttyACM0         'PX4 FMU v2.x'
--- Enter port index or full name: 1
--- Miniterm on /dev/ttyACM0  9600,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
�(␂␀␏'␏'␀␀␀␀␁␀|��␕6␁
```

### Node logs

```
... logging to /home/pkolomiets/.ros/log/10acc882-c535-11e7-8699-54ee75ba8578/roslaunch-pkolomiets-lpr-11395.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pkolomiets-lpr:34713/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /hector_trajectory_server_global/source_frame_name: base_link
 * /hector_trajectory_server_global/target_frame_name: map
 * /hector_trajectory_server_global/trajectory_publish_rate: 0.25
 * /hector_trajectory_server_global/trajectory_topic: trajectory_map
 * /hector_trajectory_server_global/trajectory_update_rate: 50.0
 * /hector_trajectory_server_local/source_frame_name: base_link_odom
 * /hector_trajectory_server_local/target_frame_name: odom
 * /hector_trajectory_server_local/trajectory_publish_rate: 0.25
 * /hector_trajectory_server_local/trajectory_topic: trajectory_odom
 * /hector_trajectory_server_local/trajectory_update_rate: 50.0
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
[... 178 lines cut ...]

### Diagnostics

```
header: 
  seq: 0
  stamp: 
    secs: 1510222328
    nsecs: 459785529
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: FCU connection
    message: Node starting up
    hardware_id: ''
    values: []
---
```

### Check ID

```
rosrun mavros checkid
```

TSC21, 2017-11-09:
This is a duplication of https://github.com/mavlink/mavros/issues/809 and https://github.com/mavlink/mavros/issues/566. Check https://github.com/mavlink/mavros/issues/809#issuecomment-329099018

closed 2017-11-09 by TSC21

pavloblindnology, 2017-11-17:
In my case, restarting the corresponding driver (using `modprobe`) did help.
So, I don't need to restart companion computer anymore - yoohoo!
To find and restart the driver see, for example, this [link](https://superuser.com/questions/572034/how-to-restart-ttyusb).

TSC21, 2017-11-17:
Seems like a good solution. Thanks!

Pedro-Roque, 2017-11-17:
On my case. restarting the computers or autopilot did not fix. I will check restarting modprobe!

Pedro-Roque, 2017-11-17:
@pavloblindnology worked for us, routine:
1. disconnect usb
2. sudo modprobe -r usbhid cdc_acm
3. then sudo modprobe usbhid cdc_acm
4. reconnect USB
5. roslaunch mavros px4.launch

TSC21, 2017-11-17:
@Pedro-Roque can you make a PR for a `Troubleshooting` section to MAVROS Readme and add the above?

Pedro-Roque, 2017-11-17:
@TSC21 Yes I can! :)

vooon, 2017-11-17:
I'm not sure that you need explicit modprobe when you plugin in usb cable. Should be done automagically. PnP: Plug and Pray :)
And that method only feasible for USB-UART, because SoC UART modules often built-in.

pavloblindnology, 2017-11-17:
@vooon Replugging USB doesn't help, only hardcore.

vooon, 2017-11-18:
@pavloblindnology you tried do only `modprobe -r` part (e.g. skip step 3)?

pavloblindnology, 2017-11-20:
@vooon For me the next steps work:

1. sudo modprobe -r cdc_acm
2. sudo modprobe cdc_acm

`modprobe -r` removes the kernel, so you don't have the corresponding port at all (`FCU: DeviceError:serial:open: No such file or directory` by mavros).

vooon, 2017-11-20:
Usually autodetect should load driver, but seems that it do not work. Maybe when you manually unload driver autodetect stop working?

pavloblindnology, 2017-11-20:
@vooon Cannot tell you. But surely no autodetection happens in my case.