mavlink/mavros #857: How to interpret cam_imu_stamp?

Changliu52, 2017-11-12:
Hi @mhkabir,
I was trying to do camera-imu hardware sync for vins_mono.
I made sure the camera is in external trigger mode and px4 (pixracer 1.6.5) enabled camera trigger following https://dev.px4.io/en/tutorials/camera_trigger.html. I also carefully modified the camera driver to match the image frame sequence with the cam_imu_stamp sequence, then overwrite the image frame stamp by the cam_imu_stamp + 0.5 exposure time. (https://github.com/Changliu52/ueye_cam)

However, from the test, the odometry is much worse than the unsyncronized setup with exactly the same hardware, in fact, most time failed. I suspect there must be something wrong fundamentally. 

Please give some advice.

Thanks

Changliu52, 2017-11-13:
In your ueye nodelet example: https://github.com/ProjectArtemis/ueye_cam/blob/artemis-devel/src/ueye_cam_nodelet.cpp#L1230, the way you over write the timestamp really doesn't make sense. It looks like you are accumulating the cam_imu_stamp in each iteration. This doesn't seem to work for me either. I am really confused....

@mhkabir Please help.

Changliu52, 2017-11-13:
My bad... at startup, there was always one image in the ueye buffer, which causes one frame shift for sequence matching. I can confirm the mavros sync was perfectly done!

closed 2017-11-13 by Changliu52

mhkabir, 2017-11-16:
:) Awesome!

Changliu52, 2017-11-20:
https://youtu.be/afJtmZv1gT8 a quick link to the performance. Really great performance!