mavlink/mavros #862: WaypointPush service

tuuzdu, 2017-11-14:
Hi!
I use ros kinetic and sitl px4.
Make mission by python script and call service:
```
rospy.wait_for_service('mavros/mission/push')
    try:
        service = rospy.ServiceProxy('mavros/mission/push', WaypointPush)
        service(0, waypoints)
    except rospy.ServiceException, e:
        print('Service call failed: {0}'.format(e))
```
_waypoints_ (in service call) type is [WaypointList](http://docs.ros.org/api/mavros_msgs/html/msg/WaypointList.html).
If _waypoints_ contains only one item mission loads normal.
If _waypoints_ contains more items, I get "WP: upload failed: command is not supported"

closed 2017-11-14 by tuuzdu

vooon, 2017-11-15:
It should be `Waypoint[]`.

tuuzdu, 2017-11-15:
Yes, right. It was `Waypoint[]`. But I have problem with waypoint cmd `MAV_CMD_NAV_RETURN_TO_LAUNCH`, which equals 20. It was reason of `WP: upload failed: command is not supported`. 
I didnt decide, why this happend.

vooon, 2017-11-15:
FCU may support some subset of nav commands. But strange that it failed on RTL, try to use MissionPlanner first.

tuuzdu, 2017-11-15:
I use qgroundcontrol and it works normal. Only when I send mission data by mavros, I have problem with `MAV_CMD_NAV_RETURN_TO_LAUNCH`. If you want, I can send any logs.

TSC21, 2017-11-15:
I would guess that it could be related to https://github.com/mavlink/mavros/pull/814 but the RTL command/mode does not require RC/GS connection - it even triggers as a failsafe.

vooon, 2017-11-16:
Logs would be nice to see. Did you tried export mission at QGC (i hope the still may export in old format - text table, json not yet supported) and then push using `mavwp`?

tuuzdu, 2017-11-16:
I followed your advice and downloaded mission (uploaded by qgroundcontrol, mission contain rtl waypoint) with `rosrun mavros mavwp show`. I saw  
`3 |  No  | Yes  | MIS (2) |    RTL (20)   | 0.0  | 0.0 | 0.0 | 0.0 |      0.0      |      0.0      |  0.0  `
My rtl waypoint have another frame (like other) `MAV_FRAME_GLOBAL_RELATIVE_ALT = 3`
When I set rtl waypoint frame` MAV_FRAME_MISSION = 2` mission uploads and works.
Is it normal?

vooon, 2017-11-16:
Seems that RTL should be in MISSION frame. And this looks logical, because RTL do not rely on altitude given in param7.