mavlink/mavros #866: lunar mavros shell script error

fnoop, 2017-11-18:
### Issue details

```
[dev] [mav@maverick-up ~/software/maverick]$sudo /bin/bash  /srv/maverick/software/ros/lunar/lib/mavros/install_geographiclib_datasets.sh
+ [[ 0 != 0 ]]
/srv/maverick/software/ros/lunar/lib/mavros/install_geographiclib_datasets.sh: line 18: syntax error in conditional expression
/srv/maverick/software/ros/lunar/lib/mavros/install_geographiclib_datasets.sh: line 18: syntax error near `or'
/srv/maverick/software/ros/lunar/lib/mavros/install_geographiclib_datasets.sh: line 18: `	if [[ -d "/usr/share/GeographicLib/$dir" or -d "/usr/local/share/GeographicLib/$dir" ]]; then'
```

### MAVROS version and platform

Mavros: 0.21.2-0stretch-20171012-121033-0800
ROS: Lunar
Debian: Stretch (Ubilinux 4.0)

---
Replacing the double brackets with single brackets and replacing 'or' with '-o' works.

vooon, 2017-11-18:
It is already fixed in upstream, and released too.

closed 2017-11-18 by vooon

fnoop, 2017-11-19:
OK, the lunar packages I installed yesterday still had the problem.