mavlink/mavros #867: install_geographiclib_datasets.sh fails on empty directories

fnoop, 2017-11-20:
### Issue details

If download of the geographiclib datasets fails for any reason (like it hangs downloading or can't reach), it leaves empty subdirectories of /usr/share/GeographicLib.  The bash script only checks for existence of these directories, not of the datasets within so any subsequent runs of the script do nothing even if the datasets are not present.

Instead, it should check for the actual files.  Change:
```
	if [[ -d "/usr/share/GeographicLib/$dir" or -d "/usr/local/share/GeographicLib/$dir" ]]; then
		return
	fi
```
 to (eg.)
```
	files=$(shopt -s nullglob dotglob; echo /usr/share/GeographicLib/$dir/$model* /usr/local/share/GeographicLib/$dir/$model*)
	if (( ${#files} )); then
		return
	fi
```

### MAVROS version and platform

Mavros: master
ROS: Lunar
Ubuntu: Debian 9 / Ubuntu 16.04

TSC21, 2017-11-20:
Thanks for pointing that @fnoop. Would you mind making a PR with this change? Thanks

fnoop, 2017-11-20:
Sure, PR #868 raised

TSC21, 2017-11-29:
Closed by #868

closed 2017-11-29 by TSC21