mavlink/mavros #869: Failing to parse /mavros/mocap/pose message to pixhawk?

jannsta1, 2017-11-20:
### Issue details

I'm trying to get external position estimation working via mavros. I think there is a problem parsing the /mavros/mocap/pose data to the local_position_estimator on the pixhawk controller as I don't get any rosinfo messages published from [lpe] and the main pixhawk light remains flashing blue rather than turning green. I was following https://dev.px4.io/en/ros/external_position_estimation.html but no doubt I haven't set something up properly.

As you can see from the diagnostics the serial comms seem to work OK although there are some warnings on particular messages on startup that could indicate a problem - but looking at other issue reports this seems like a typical output. There are also some warnings about unexpected commands (520 and 176) which I've included in the node log.

As far as I can tell ATT_EXT_HDG_M is set to 2 and the relevant parameters are set:
```
drone@rp2:~/px4_ws/src/mavros$ rosparam get /mavros/fake_gps/use_mocap
true
drone@rp2:~/px4_ws/src/mavros$ rosparam get /mavros/fake_gps/use_vision
false
drone@rp2:~/px4_ws/src/mavros$ rosparam get /mavros/mocap/use_pose
true
drone@rp2:~/px4_ws/src/mavros$ rosparam get /mavros/mocap/use_tf
false
```

and I've tried sending /mavros/mocap/pose at different rates:
```
drone@rp2:~/px4_ws/src/mavros$ rostopic echo -n5 /mavros/mocap/pose
header: 
  seq: 3081
  stamp: 
    secs: 1511178635
    nsecs: 789153099
  frame_id: mavros/mocap/pose
pose: 
  position: 
    x: 0.00353349181057
    y: 0.00889526472434
    z: 0.00666116065408
  orientation: 
    x: 0.00393101160921
    y: 0.00484008427282
    z: 0.00403871044368
    w: 0.999972404394
---
header: 
  seq: 3082
  stamp: 
    secs: 1511178635
    nsecs: 899713039
  frame_id: mavros/mocap/pose
pose: 
  position: 
    x: 0.000226181153103
    y: 0.00896079078765
    z: 0.00835343181798
  orientation: 
    x: 0.00101887339043
    y: 0.00473004082867
    z: 0.00309594826198
    w: 0.999983501721
---
header: 
  seq: 3083
  stamp: 
    secs: 1511178636
    nsecs:  10835886
  frame_id: mavros/mocap/pose
pose: 
  position: 
    x: 0.00413232438773
    y: 0.00285399702229
    z: 0.00191113504611
  orientation: 
    x: 0.000851472984996
    y: 0.003993808153
    z: 0.00224211881842
[... 480 lines cut ...]
      - 
        key: Mean dt (ms)
        value: 0.026988
      - 
        key: Last system time (s)
        value: 55.806099000
      - 
        key: Time offset (s)
        value: 1511179134.491588593
---

```

### Check ID

```
rosrun mavros checkid:

OK. I got messages from 1:1.
---
Received 6829 messages, from 2 addresses
sys:comp   list of messages
  1:214   0
  1:1     32, 1, 2, 245, 36, 230, 147, 105, 74, 76, 141, 77, 111, 241, 140, 83, 85, 87, 0, 30, 31
```

TSC21, 2017-11-20:
Do you have `SYS_MC_EST_GROUP` set to `1`?

jannsta1, 2017-11-20:
Yes, ```SYS_MC_EST_GROUP``` is indeed set to ```1``` or ```'local_position_estimator, attitude_estimator_q'``` in QGC speak.

jannsta1, 2017-11-21:
After sorting out gcs link via network (this had been incorrectly implemented by me but I hadn't noticed as I was getting data through a radio telemetry link) I now do get a message through from lpe:
```
[ INFO] [1511258248.847161373]: FCU: [lpe] mocap position init:  0.00,  0.01,  0.75 m 
[ INFO] [1511258248.899148351]: FCU: [lpe] reinit P (0, 0) negative
```
Just looking at other issues with this symptom. Changes to the LPE parameters seem to result in a solution. My LPE params are all at default except for ```LPE_FUSION``` which is ```1``` as I read its best to ignore barometer. I need to look into these parameters in order to understand their meaning before I start changing them.

I did notice that there is a small but constant vz value as per the plot below - could this indicate the problem?

[image]

TSC21, 2017-11-21:
> I did notice that there is a small but constant vz value as per the plot below - could this indicate the problem?

If you are fusing baro, that's perfectly normal.

jannsta1, 2017-11-21:
I wasn't fusing baro but I guess as the offset is not running away it isn't a concern?

I have tried a spread of values for ```LPE_VIC_P``` and ```LPE_PN_V```. The only thing I have really observed is that when I reduce ```LPE_VIC_P``` the ```FCU: [lpe] reinit P (0, 0) negative``` warning message becomes periodic rather than just once on initialisation.

Now I'm confused if lpe is initialising in the case with the case with the default ```LPE_VIC_P``` (the case with a single reinit warning message) or not. Would I expect to see a message like ```[lpe] fuse gps: 1, flow: 0, vis_pos: 0, vis_yaw: 0, land: 0, pub_agl_z: 0, flow_gyro: 0``` if lpe initialises properly? (I don't). 

Any other suggestions on what to look into in order to diagnose this?

jannsta1, 2017-11-27:
actually, if I increase LPE_VIC_P to 0.001 (as described in https://github.com/PX4/Firmware/issues/7884) then ```FCU: [lpe] mocap position init: -0.00,  0.00,  0.10 m ``` is printed in the terminal and looks like all is OK.

However, the main LED on the pixhawk doesn't turn green when I transition from STABILIZED to POSCTL or OFFBOARD which seems to contradict the 'first flight' section in: https://dev.px4.io/en/ros/external_position_estimation.html. 

It seems like offboard position control is working as if I stop broadcasting the ~/mocap/pose message it is no longer possible to transition from STABILIZED to POSCTL. When I transition to POSCTL with ~/mocap/pose present the main LED is solid blue.

Not sure if the LED not becoming green is an issue @TSC21? / if more relevant to the px4 repo?

TSC21, 2017-11-27:
The LED does not change to green when yo change flight mode: it should automatically change to green when it gets a position lock, meaning that there is a position estimation. And that works the same way to LPE or EKF2.

jannsta1, 2017-11-27:
I was working on the assumption that position lock is required in order to transition to position mode? 

(Thus if the pixhawk transitions to POSCTL then I assume the position lock condition is satisfied and by extension the main LED should be green)

jannsta1, 2017-11-27:
@TSC21. To me, the documentation at https://docs.px4.io/en/flight_modes/offboard.html suggests that position lock is required to transition to OFFBOARD mode (i.e. the LED should be green before this transition occurs). I refer to the text that says - _'This mode requires position or pose/attitude information - e.g. GPS, optical flow, visual-inertial odometry, mocap, etc'_.

TSC21, 2017-11-27:
And that is correct.

hamishwillee, 2017-11-27:
>  - 'This mode requires position or pose/attitude information - e.g. GPS, optical flow, visual-inertial odometry, mocap, etc'.

Thanks very much for that text in the user guide @TSC21 :-)

jannsta1, 2017-11-27:
@TSC21  - so if I have a blue flashing* main LED and can transition (either via RC or mavros service) from STABILIZED to either POSCTL or OFFBOARD then this is unintended functionality? (sorry to keep pestering just not clear to me if this is a bug or something wrong with my setup). 

*a subsequent arm command causes the main LED to turn solid blue

TSC21, 2017-11-27:
Something is not right on that setup. You shouldn't be able to change to POSCTL without a position lock (solid green after arming).

jannsta1, 2017-11-28:
Reflashed PX4 sw today (tried various releases) and set all parameters back to default and still no green LED but still able to transitions to POSCTL when I broadcast /mavros/mocap/pose.

Noticed that in the PX4 source that green LED conditions are global_position_valid && home_position_valid.

In my setup there don't appear to be any ROS topics published on /mavros/global_position/*. Nor does home position get set (periodically see ```HP: requesting home position```) - @TSC21 would these be expected to be present if I had lpe configured to operate properly?

TSC21, 2017-11-28:
That basically means that there is no position lock. Do you have the FAKEGPS parameter set?

jannsta1, 2017-11-28:
Ah yeah, if I set ```LPE_FAKE_ORIGIN``` to 1 then the LED will turn green.

Will leave for you to close in case the fact that the vehicle arms without ```LPE_FAKE_ORIGIN``` for non-gps setups is considered a problem.

Thanks @TSC21 for your help!!

TSC21, 2017-11-28:
The vehicle should be able to arm and takeoff even if there is no position lock. It should not change to position dependent modes in that case though.

closed 2017-11-28 by TSC21

jannsta1, 2017-12-01:
@TSC21, sorry it wasn't clear in my previous message but I think that in lpe mode it is possible to move to POSCTL & OFFBOARD modes & arm the copter without the position lock, or at least without the LED turning green.

This is only the case when you send either the tf or pose information to ~/mocap/* and LPE_FAKE_ORIGIN is not set to zero. 

Another thing I noticed is that when I set LPE_FAKE_ORIGIN to 1 and px4 acquires a position lock the LED remains green even when I stop my roscore. I would have guessed that the LED should return to blue once the position lock is lost.

Attached is a rosbag of the transition from STABILIZED to POSCTL I believe. I just cloned mavros and px4 repos so the only things I have changed are the parameters.
 
[POSCTL_with_no_global_pos_2017-12-01-09-37-07.bag.zip](https://github.com/mavlink/mavros/files/1521927/POSCTL_with_no_global_pos_2017-12-01-09-37-07.bag.zip)

TSC21, 2017-12-01:
We need logs and not bags to help you out. I still it weird that is happening. I currently don't have bandwidth to reproduce what you are stating. I don't have a Dropix, but I never saw that behavior with other HW.

TSC21, 2017-12-01:
> Another thing I noticed is that when I set LPE_FAKE_ORIGIN to 1 and px4 acquires a position lock the LED remains green even when I stop my roscore. I would have guessed that the LED should return to blue once the position lock is lost.

You need to check if mocap reaches timeout or not. Check the statustext.

jannsta1, 2017-12-03:
@TSC21 - attached is a .ulg log of the behaviour if this is any use? It's not holding me up now that I know the steps to get a green LED but thought it seemed worth identifying the reason for this?

> You need to check if mocap reaches timeout or not. Check the statustext.

Thanks for the diagnostic tip - alas, I do get the mocap timeout text from lpe. I've also tried leaving the pixhawk unit powered for several minutes and the LED seems to remain green.

[POSCTL_with_no_global_pos_2017-12-03.ulg.zip](https://github.com/mavlink/mavros/files/1525077/POSCTL_with_no_global_pos_2017-12-03.ulg.zip)