mavlink/mavros #872: MAV_SYS_ID is only working if 1.

AlexisTM, 2017-11-24:
### Issue details

* If MAV_SYS_ID and tgt_system = 1: Everything works
* If MAV_SYS_ID and tgt_system = 82: Data flows yet the state stays not connected.
* Data flows means I receive data on topics `/mavros/local_position/local` or the imu
* State stays not connected means the topic `/mavros/state` reports `connected: False`

### MAVROS version and platform

Mavros: 0.21.3
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[x] PX4

Version: 1.6.5

### Diagnostics

```
rostopic echo -n1 /diagnostics

header: 
  seq: 59
  stamp: 
    secs: 1455208301
    nsecs: 681187693
  frame_id: ''
status: 
  - 
    level: 1
    name: odroid82: FCU connection
    message: not connected
    hardware_id: /dev/ttyUSB0:921600
    values: 
      - 
        key: Received packets:
        value: 52862
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 222
      - 
        key: Tx sequence number:
        value: 10
      - 
        key: Rx total bytes:
        value: 2261764
      - 
        key: Tx total bytes:
        value: 40108
      - 
        key: Rx speed:
        value: 23808.000000
      - 
        key: Tx speed:
        value: 570.000000
  - 
    level: 0
    name: odroid82: GPS
    message: 3D fix
    hardware_id: /dev/ttyUSB0:921600
    values: 
      - 
        key: Satellites visible
        value: 9
      - 
        key: Fix type
        value: 3
      - 
        key: EPH (m)
        value: 1.02
      - 
        key: EPV (m)
        value: 2.07
  - 
    level: 2
    name: odroid82: Heartbeat
    message: No events recorded.
    hardware_id: /dev/ttyUSB0:921600
    values: 
      - 
        key: Heartbeats since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Vehicle type
        value: Generic micro air vehicle
      - 
        key: Autopilot type
        value: Generic autopilot
      - 
        key: Mode
        value: ''
      - 
        key: System status
        value: Uninit
  - 
    level: 0
    name: odroid82: System
    message: Normal
    hardware_id: /dev/ttyUSB0:921600
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 61.9
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 0
    name: odroid82: Battery
    message: Normal
    hardware_id: /dev/ttyUSB0:921600
    values: 
      - 
        key: Voltage
        value: 24.38
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 88.0
  - 
    level: 0
    name: odroid82: Time Sync
    message: Normal
    hardware_id: /dev/ttyUSB0:921600
    values: 
      - 
        key: Timesyncs since startup
        value: 91
      - 
        key: Frequency (Hz)
        value: 9.791079
      - 
        key: Last dt (ms)
        value: -4.602662
      - 
        key: Mean dt (ms)
        value: -0.096830
      - 
        key: Last system time (s)
        value: 237.806095000
      - 
        key: Time offset (s)
        value: 1455208063.785467863
---
```

### Check ID

```
rosrun mavros checkid

ERROR. I got 0 addresses, but not your target 0:1

---
Received 0 messages, from 0 addresses
sys:comp   list of messages
```

vooon, 2017-11-25:
SYS ID should be > 0, because 0 means broadcast.
Please post dump of `checkid` when you setup sys-id=82.
Also post node logs (we do not need full log, only lines after plugins loaded).

AlexisTM, 2017-11-25:
@vooon The dump in the post is when the MAV_SYS_ID on the Pixhawk is 82.

vooon, 2017-11-25:
Nope, checkid ids=0:1 ⇒ system 0.

AlexisTM, 2017-11-25:
Then that should be the problem; I can assure you that I never tried the ID 0 on MAV_SYS_ID, but I did try target_system to 0 to try to contact it.

TSC21, 2017-12-23:
@AlexisTM any updates?

AlexisTM, 2017-12-23:
I did not retry yet as lot of company stuff happened in between. I will retry  at the same time as the 1.7.x firmware test after my holidays (3rd of December) ;)

AlexisTM, 2018-04-04:
@Ekci I am using `mav_sys_id` to 1 for all the copters and never use multiple data stream at the same time. :/

Ekci, 2018-04-04:
Ok, I have found out that chaning the ```tgt_system``` argument in [https://github.com/mavlink/mavros/blob/master/mavros/launch/px4.launch](px4.launch) to correct value. It doesn't look right, with wrong argument it works but not fully, is there a reason to require it? I have to dig in.

AlexisTM, 2021-06-30:
I am closing this as it is very likely outdated (and I am not in possession of a copter for testing).

closed 2021-06-30 by AlexisTM