mavlink/mavros #873: Error build package when install mavros

lp02781, 2017-11-27:
i have some problems, when i try to install mavros on ubuntu 16.04 ROS kinetic, there error in build package like in ss. what must i do??

[image]

TSC21, 2017-11-27:
Are you running a VM? That seems a problem with low RAM or swap space. This is not an issue directly related with MAVROS. Try increasing the swap space or follow https://www.cyberciti.biz/faq/linux-add-a-swap-file-howto/

closed 2017-11-27 by TSC21

lp02781, 2017-11-28:
thank for your advice

lp02781, 2017-11-28:
i'm not use VM, i'm using dual boot windows 8 and ubuntu 16.04, i'm installing ubuntu not as swap partition, but as extent partition, that give effect for my ubuntu??

vooon, 2017-11-28:
Please show `free -h` and `lscpu` (copy text and place inside \`\`\` code block).

vooon, 2017-11-28:
4 GiB of RAM is too low to run 4 GCC simultaneously (yes C++ take lots of RAM, ~1-1.5G for each process).
You should add swap partition or file, it is necessary when RAM < 6-8G (and still recommended to have small area even if you have more ram).

Quick way to add swap file:
```sh
sudo fallocate -l 6G /swapfile
sudo mkswap /swapfile
sudo swapon /swapfile
echo '/swapfile none swap sw 0 0' | sudo tee -a /etc/fstab
```

But swapping may cause that system will be very slow, limitation number of parallel jobs may help.
Try add flag `-j3` to `catkin build`.