mavlink/mavros #877: setpoint_velocity doesn't work

pavloblindnology, 2017-11-29:
Due to [ardupilot #7324](https://github.com/ArduPilot/ardupilot/issues/7324) setpoint_velocity doesn't work, because neither option is selected in ardupilot [here](https://github.com/ArduPilot/ardupilot/blob/68800a52f5c73fc0c0b4b94357a3198ba16bf60d/APMrover2/GCS_Mavlink.cpp#L1141) or [here](https://github.com/ArduPilot/ardupilot/blob/68800a52f5c73fc0c0b4b94357a3198ba16bf60d/APMrover2/GCS_Mavlink.cpp#L1220)
https://github.com/mavlink/mavros/blob/6f5bd5a1a67c19c2e605f33de296b1b1be9d02fc/mavros/src/plugins/setpoint_velocity.cpp#L82

TSC21, 2017-11-29:
@pavloblindnology that should be something to fix on Ardupilot side. Here you can just use `setpoint_raw`.

khancyr, 2017-11-29:
Can be closed here, it is only a problem on ardupilot side ... correction is coming

closed 2017-11-29 by TSC21