mavlink/mavros #878: MODE: Unsupported FCU with offb_node

SarahMars, 2017-11-30:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details
Hello,
I am trying to do offboard control with a pixracer. I want to fly it indoor so it has px4flow sensor and I am using telemetry modules(mRo SiK Telemetry Radio 915MHz). I have already tried offb_node simulation with gazebo and had no problem with it. Although I can't do it with an actual copter.

First, I opened Qgroundcontrol and I could see it connected with quadcopter through telemetry.
Second, I launched  MAVROS with
`roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"`
Third, with a new terminal, I typed 
`rosrun mavros offb_node`
and then motors didn't move and nothing happened.
I just got `MODE: Unsupported FCU`
Am I doing some steps wrong?

Also when I run mavros_node,  MAVROS will shut down even I didn't change node name. (I checked offb_node and mavros_node have different name inside source code.)

`[ WARN] [1512076026.741290904]: Shutdown request received.
[ WARN] [1512076026.741376126]: Reason given for shutdown: [new node registered with same name]
================================================================================REQUIRED process [mavros-2] has died!
process has finished cleanly
log file: /home/sarah/.ros/log/6692cc4a-d612-11e7-905c-606c66107b1f/mavros-2*.log
Initiating shutdown!
================================================================================`

Thank you,
Sarah

### MAVROS version and platform

Mavros: 0.20.0
ROS: Lunar
Ubuntu: 16.04

### Autopilot type and version
PX4
Version: I don't know which version it is

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch

[ INFO] [1512076026.757365212]: FCU URL: udp://:14540@127.0.0.1:14557
[ WARN] [1512076026.758281701]: init: message from ASLUAV, MSG-ID 201 ignored! Table has different entry.
[ WARN] [1512076026.758494189]: init: message from autoquad, MSG-ID 150 ignored! Table has different entry.
[ WARN] [1512076026.758522671]: init: message from autoquad, MSG-ID 152 ignored! Table has different entry.
[ WARN] [1512076026.758817391]: init: message from matrixpilot, MSG-ID 150 ignored! Table has different entry.
[ WARN] [1512076026.758846686]: init: message from matrixpilot, MSG-ID 151 ignored! Table has different entry.
[ WARN] [1512076026.758873514]: init: message from matrixpilot, MSG-ID 152 ignored! Table has different entry.
[ WARN] [1512076026.758900131]: init: message from matrixpilot, MSG-ID 153 ignored! Table has different entry.
[ WARN] [1512076026.758925669]: init: message from matrixpilot, MSG-ID 155 ignored! Table has different entry.
[ WARN] [1512076026.758953076]: init: message from matrixpilot, MSG-ID 156 ignored! Table has different entry.
[ WARN] [1512076026.758979508]: init: message from matrixpilot, MSG-ID 157 ignored! Table has different entry.
[ WARN] [1512076026.759036606]: init: message from matrixpilot, MSG-ID 158 ignored! Table has different entry.
[ WARN] [1512076026.759066559]: init: message from matrixpilot, MSG-ID 170 ignored! Table has different entry.
[ WARN] [1512076026.759093780]: init: message from matrixpilot, MSG-ID 171 ignored! Table has different entry.
[ WARN] [1512076026.759119528]: init: message from matrixpilot, MSG-ID 172 ignored! Table has different entry.
[ WARN] [1512076026.759172047]: init: message from matrixpilot, MSG-ID 173 ignored! Table has different entry.
[ WARN] [1512076026.759203431]: init: message from matrixpilot, MSG-ID 174 ignored! Table has different entry.
[ WARN] [1512076026.759229775]: init: message from matrixpilot, MSG-ID 175 ignored! Table has different entry.
[ WARN] [1512076026.759256874]: init: message from matrixpilot, MSG-ID 176 ignored! Table has different entry.
[ WARN] [1512076026.759283166]: init: message from matrixpilot, MSG-ID 177 ignored! Table has different entry.
[ WARN] [1512076026.759308999]: init: message from matrixpilot, MSG-ID 178 ignored! Table has different entry.
[ WARN] [1512076026.759334205]: init: message from matrixpilot, MSG-ID 179 ignored! Table has different entry.
[ WARN] [1512076026.759360751]: init: message from matrixpilot, MSG-ID 180 ignored! Table has different entry.
[ WARN] [1512076026.759387069]: init: message from matrixpilot, MSG-ID 181 ignored! Table has different entry.
[... 182 lines cut ...]
        value: "0"
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
        key: "Last dt (ms)"
        value: "0.000000"
      - 
        key: "Mean dt (ms)"
        value: "0.000000"
      - 
        key: "Last system time (s)"
        value: "0.000000000"
      - 
        key: "Time offset (s)"
        value: "0.000000000"
---

```

### Check ID

```
rosrun mavros checkid
```

vooon, 2017-12-01:
```
    name: "mavros: FCU connection"
    message: "not connected"
```

TSC21, 2017-12-01:
@SarahMars I don't think your setup is right. You are using QGC as intermediary between MAVROS and the FCU. You should use MAVROS instead as intermediary between the FCU and QGC. So you should set the FCU url to your serial link, and the QGC URL (on the launch file) to something like `udp://:14550@localhost:14557`

TSC21, 2017-12-01:
Also, since you are using a telemetry link, make sure to set the `SYS_COMPANION` computer to 57600, while the serial link to `/dev/ttyACM0:57600`.

SarahMars, 2017-12-01:
@TSC21 
Thank you for your reply. I checked SYS_COMPANION on QGC and it was OSD (57600 baud 8N1).

If I change this line `<arg name="gcs_url" default=""/> ` in my px4.launch file to `<arg name="gcs_url" default="udp://:14550@127.0.0.1:14557"/> ` , and then when I run `rosrun mavros px4.launch`
I get this error.

`[ INFO] [1512149289.750564794]: serial0: device: /dev/ttyACM0 @ 57600 bps
[FATAL] [1512149289.750884993]: FCU: DeviceError:serial:open: No such file or directory
================================================================================REQUIRED process [mavros-2] has died!
process has finished cleanly
log file: /home/sarah/.ros/log/ffbd1cfe-d6bc-11e7-aae1-606c66107b1f/mavros-2*.log
Initiating shutdown!
================================================================================
[mavros-2] killing on exit
[rosout-1] killing on exit
`

Here is my px4.launch code just in case.
`<launch>
	
	

	<arg name="fcu_url" default="/dev/ttyACM0:57600" />
	<arg name="gcs_url" default="udp://:14550@127.0.0.1:14557"/> 

	<arg name="tgt_system" default="1" />
	<arg name="tgt_component" default="1" />
	<arg name="log_output" default="screen" />

	<include file="$(find mavros)/launch/node.launch">
		<arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
		<arg name="config_yaml" value="$(find mavros)/launch/px4_config.yaml" />

		<arg name="fcu_url" value="$(arg fcu_url)" />
		<arg name="gcs_url" value="$(arg gcs_url)" />
		<arg name="tgt_system" value="$(arg tgt_system)" />
		<arg name="tgt_component" value="$(arg tgt_component)" />
		<arg name="log_output" value="$(arg log_output)" />
	</include>
</launch>`

TSC21, 2017-12-01:
> `FCU: DeviceError:serial:open: No such file or directory`

If not `ACM0`, can be `USB0`. Check on `/dev/` for `ACM0` or `USB0`.

SarahMars, 2017-12-01:
@TSC21 
I got this after changing it to USB0. By the way where is /dev/ located at?
`[ INFO] [1512150367.312024209]: serial0: device: /dev/ttyUSB0 @ 57600 bps
[FATAL] [1512150367.312338661]: FCU: DeviceError:serial:open: Device or resource busy
REQUIRED process [mavros-2] has died!
process has finished cleanly
log file: /home/sarah/.ros/log/8203b0c2-d6bf-11e7-aae1-606c66107b1f/mavros-2*.log
Initiating shutdown!
`

TSC21, 2017-12-01:
You cannot have QGC to automatically connect to the FCU. Go to QGC settings and disable the option to connect to the `Pixhawk` and the `SiK Radio`.

TSC21, 2017-12-01:
[image]

TSC21, 2017-12-01:
`By the way where is /dev/ located at?`

On your system's root.

SarahMars, 2017-12-01:
@TSC21 
After disabling pixhawk and SiK radio, I got this

```[ INFO] [1512150802.265587539]: serial0: device: /dev/ttyUSB0 @ 57600 bps
[ INFO] [1512150802.266750092]: GCS URL: udp://:14550@127.0.0.1:14557
[ERROR] [1512150802.266814919]: serial0: receive: End of file
terminate called after throwing an instance of 'std::system_error[ INFO] [1512150802.267336995]: udp1: Bind address: 0.0.0.0:14550
'
  what():  Resource deadlock avoided
================================================================================REQUIRED process [mavros-2] has died!
process has died [pid 11190, exit code -6, cmd /home/sarah/catkin_ws5/devel_isolated/mavros/lib/mavros/mavros_node __name:=mavros __log:=/home/sarah/.ros/log/8541fe1e-d6c0-11e7-aae1-606c66107b1f/mavros-2.log].
log file: /home/sarah/.ros/log/8541fe1e-d6c0-11e7-aae1-606c66107b1f/mavros-2*.log
Initiating shutdown!
```

TSC21, 2017-12-01:
https://github.com/mavlink/mavros/blob/master/mavros/README.md#troubleshooting

SarahMars, 2017-12-01:
Sorry I'm still searching for how to check /dev/ (I'm new to linux)

TSC21, 2017-12-01:
> Sorry I'm still searching for how to check /dev/ (I'm new to linux)

`cd /dev`. Lots of docs online. Start with for example: https://www.ibm.com/developerworks/library/l-lpic1-103-1/index.html

SarahMars, 2017-12-01:
Thank you for the example website, I'll take a look at it.

I've tried https://github.com/mavlink/mavros/blob/master/mavros/README.md#troubleshooting , now it gave me
```[ INFO] [1512151658.944483415]: serial0: device: /dev/ttyUSB0 @ 57600 bps
[FATAL] [1512151658.944907999]: FCU: DeviceError:serial:open: No such file or directory
================================================================================REQUIRED process [mavros-2] has died!
process has finished cleanly
log file: /home/sarah/.ros/log/83e340e4-d6c2-11e7-aae1-606c66107b1f/mavros-2*.log
Initiating shutdown!
================================================================================```

cd/dev/ls is like this

![image](https://user-images.githubusercontent.com/32495711/33496732-5de473dc-d699-11e7-94fb-654bdb1f4494.png)

TSC21, 2017-12-01:
You have `ACM0` available.

SarahMars, 2017-12-01:
Oh I `ttyACM0`, I got it.

Thank you so much for your help, I can now connect to the FCU!
``` 
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "/dev/ttyACM0:57600"
```

TSC21, 2017-12-01:
OK. QGC got the connection also?

SarahMars, 2017-12-01:
Oh wait, I wasn't opening QGC while running the command.

```
[ INFO] [1512153362.161605754]: serial0: device: /dev/ttyACM0 @ 57600 bps
[ INFO] [1512153362.162249010]: GCS URL: udp://:14550@127.0.0.1:14557
[ INFO] [1512153362.162880156]: udp1: Bind address: 0.0.0.0:14550
[ INFO] [1512153362.163025246]: udp1: Remote address: 127.0.0.1:14557
[FATAL] [1512153362.163568876]: GCS: DeviceError:udp:bind: Address already in use
```

TSC21, 2017-12-01:
try `udp-b://127.0.0.1:14555@14550`

SarahMars, 2017-12-01:
It worked! Thank you so much!

Although I could get connection when the USB cable is connected, how am I supposed to get connection when I want to fly it?

TSC21, 2017-12-01:
> Although I could get connection when the USB cable is connected, how am I supposed to get connection when I want to fly it?

What do you mean? Don't you have a telemetry module?

SarahMars, 2017-12-01:
@TSC21 
I was connecting both of telemetry module and USB cable. I connected both of them because it didn't connect without a USB cable. Sorry now I understood I was doing weird thing.

If I only plug in telemetry module, I get `Error: serial0: receive: End of file`. Even if I followed the steps given in [https://github.com/mavlink/mavros/blob/master/mavros/README.md#troubleshooting](url), it will still give me this error.

Also if I only connect telemetry module, it was `ttyUSB0` so I changed px4.launch file.

SarahMars, 2017-12-01:
@TSC21 
Rebooted and problem solved!
Thank you very much for all of your help!

TSC21, 2017-12-01:
No problem. Glad it is working.

closed 2017-12-01 by TSC21