mavlink/mavros #880: Not able to arm px4

SimonVleugels, 2017-12-03:
Hi,
I'm trying to control a drone from a Raspberry through mavros by using overriderc messages (pixhawk with px4 firmware). I'm using [this](https://dev.px4.io/en/ros/mavros_offboard.html) example, but instead of changing the pose of the drone (no gps), I want to use override rc messages to control the drone. I have been able to get into offboard mode, but the drone won't arm (red flashing led indicator). With the default example (using setpoint_position/local messages) I'm able to arm, but then the pixhawk shutsdown.

### MAVROS version and platform

Mavros: updated 26/11
ROS: Kinetic
Ubuntu: 16.04
Rasbian: Jessie

### Autopilot type and version

[  ] ArduPilot
[ x] PX4

Version: updated 30/11

### Diagnostics

```
rostopic echo -n1 /diagnostics
header: 
  seq: 1197
  stamp: 
    secs: 1512296801
    nsecs: 151528333
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: /dev/ttyS0:57600
    values: 
      - 
        key: Received packets:
        value: 46569
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 228
      - 
        key: Tx sequence number:
        value: 62
      - 
        key: Rx total bytes:
        value: 1495093
      - 
        key: Tx total bytes:
        value: 598083
      - 
        key: Rx speed:
        value: 862.000000
      - 
        key: Tx speed:
        value: 277.000000
  - 
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: /dev/ttyS0:57600
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 0
      - 
        key: EPH (m)
        value: Unknown
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 0
    name: mavros: Heartbeat
    message: Normal
    hardware_id: /dev/ttyS0:57600
    values: 
      - 
        key: Heartbeats since startup
        value: 1332
      - 
        key: Frequency (Hz)
        value: 1.036972
      - 
        key: Vehicle type
        value: Quadrotor
      - 
        key: Autopilot type
        value: PX4 Autopilot
      - 
        key: Mode
        value: MANUAL
      - 
        key: System status
        value: Standby
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: /dev/ttyS0:57600
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 35.4
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 0
    name: mavros: Battery
    message: Normal
    hardware_id: /dev/ttyS0:57600
    values: 
      - 
        key: Voltage
        value: 12.51
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 100.0
  - 
    level: 0
    name: mavros: Time Sync
    message: Normal
    hardware_id: /dev/ttyS0:57600
    values: 
      - 
        key: Timesyncs since startup
        value: 136
      - 
        key: Frequency (Hz)
        value: 9.999376
      - 
        key: Last dt (ms)
        value: -2.379068
      - 
        key: Mean dt (ms)
        value: -0.074346
      - 
        key: Last system time (s)
        value: 29.700256000
      - 
        key: Time offset (s)
        value: 1512296771.436703205
---
```

### Check ID

```
rosrun mavros checkid

OK. I got messages from 1:1.

---
Received 400 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 147, 230, 231, 74, 76, 77, 111, 141, 241, 83, 245, 36, 30
```

### Code

`int main(int argc, char **argv)
{
    ros::init(argc, argv, "offb_node");
    ros::NodeHandle nh;

    ros::Subscriber state_sub = nh.subscribe<mavros_msgs::State>
            ("mavros/state", 10, state_cb);
    ros::Publisher local_pos_pub = nh.advertise<geometry_msgs::PoseStamped>
            ("mavros/setpoint_position/local", 10);
    ros::ServiceClient arming_client = nh.serviceClient<mavros_msgs::CommandBool>
            ("mavros/cmd/arming");
    ros::ServiceClient set_mode_client = nh.serviceClient<mavros_msgs::SetMode>
            ("mavros/set_mode");
    ros::Publisher rc_pub = nh.advertise<mavros_msgs::OverrideRCIn>
            ( "mavros_msgs/OverrideRCIn", 1, true);

    //the setpoint publishing rate MUST be faster than 2Hz
    ros::Rate rate(20.0);

    // wait for FCU connection
    while(ros::ok() && current_state.connected){
        ros::spinOnce();
        rate.sleep();
    }

    geometry_msgs::PoseStamped pose;
    pose.pose.position.x = 0;
    pose.pose.position.y = 0;
    pose.pose.position.z = 2;

    //send a few setpoints before starting
    for(int i = 100; ros::ok() && i > 0; --i){
        local_pos_pub.publish(pose);
        ros::spinOnce();
        rate.sleep();
    }

    mavros_msgs::SetMode offb_set_mode;
    offb_set_mode.request.custom_mode = "OFFBOARD";

    mavros_msgs::CommandBool arm_cmd;
    arm_cmd.request.value = true;

    ros::Time last_request = ros::Time::now();
    mavros_msgs::OverrideRCIn rc_msg;

    while(ros::ok()){
        if( current_state.mode != "OFFBOARD" &&
            (ros::Time::now() - last_request > ros::Duration(5.0))){
            if( set_mode_client.call(offb_set_mode) &&
                offb_set_mode.response.mode_sent){
                ROS_INFO("Offboard enabled");
            }
            last_request = ros::Time::now();
        } else {
            if( !current_state.armed &&
                (ros::Time::now() - last_request > ros::Duration(5.0))){
                if( arming_client.call(arm_cmd) &&
                    arm_cmd.response.success){
                    ROS_INFO("Vehicle armed");
                }
                last_request = ros::Time::now();
            }
        }

        for(int i = 0; i < 8; i++){
            rc_msg.channels[i] = 0;
        }
        rc_msg.channels[2] = 1300;

        rc_pub.publish(rc_msg);
        ros::spinOnce();
        rate.sleep();
    }

    return 0;
}`

vooon, 2017-12-03:
What you see in mavros logs when `rosrun mavros mavsafety arm`?

SimonVleugels, 2017-12-03:
When running `rosrun mavros mavsafety arm` I get following output: 
The propellers also start rotating and he go into AUTO.LAND mode.

``` MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1512314251.613677663]: IMU: High resolution IMU detected!
[ INFO] [1512314251.614987859]: IMU: Attitude quaternion IMU detected!
[ WARN] [1512314251.615954882]: TM: Clock skew detected (-1512314235.465276957 s). Hard syncing clocks.
[ INFO] [1512314251.981568965]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1512314251.993866407]: IMU: High resolution IMU detected!
[ INFO] [1512314251.994895981]: IMU: Attitude quaternion IMU detected!
[ INFO] [1512314252.996292936]: VER: 1.1: Capabilities         0x00000000000024ef
[ INFO] [1512314252.996574392]: VER: 1.1: Flight software:     01060500 (00000000FFFFFFFF)
[ INFO] [1512314252.996768609]: VER: 1.1: Middleware software: 01060500 (00000000FFFFFFFF)
[ INFO] [1512314252.996945586]: VER: 1.1: OS software:         000000c0 (00000000FFFFFFFF)
[ INFO] [1512314252.997303239]: VER: 1.1: Board hardware:      00000011
[ INFO] [1512314252.997637924]: VER: 1.1: VID/PID:             26ac:0011
[ INFO] [1512314252.997802923]: VER: 1.1: UID:                 3034510831323737
[ WARN] [1512314253.001378359]: CMD: Unexpected command 520, result 0
```

vooon, 2017-12-04:
So no `FCU:` messages, and because motors turn on likely you successfully armed fcu.

SimonVleugels, 2017-12-04:
@vooon but then why am I not been able to arm the fcu with the mavros_msgs/OverrideRCIn messages?

SimonVleugels, 2017-12-04:
@vooon can you maybe supply me some example code which should be able to do something with the drone without using gps? Because everything I can find is using a gps.

What message types are supported for using in offboard mode and without gps? For example setpoint_velocity/cmd_vel?

closed 2017-12-24 by SimonVleugels