mavlink/mavros #884: GeographicLib_LIBRARIES set to NOTFOUND

tuloski, 2017-12-11:
Hi all. When installing from source (with catkin_make) I get the following error:

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GeographicLib_LIBRARIES (ADVANCED)
    linked by target "mavros" in directory /catkin_ws/src/mavros/mavros

I installed GeographicLib and I run the script install_geographiclib_datasets.sh. Is there somethin else I'm missing?

TSC21, 2017-12-11:
> I installed GeographicLib and I run the script install_geographiclib_datasets.sh

If you followed the src install as it is on the instructions, you wouldn't have to manually install GeographicLib in the first place, though you would still have to install the datasets. How have you installed GeographicLib? Through `apt-get`? What OS/platform are you using?

TSC21, 2017-12-11:
Also we do recommend compile using [`catkin tools`](http://catkin-tools.readthedocs.io/en/latest/verbs/catkin_build.html) instead of `catkin_make`.

tuloski, 2017-12-11:
Yeah I installed via "apt-get install geographiclib-tool". Is it the correct one? Because it's the only one I found. I'm on Ubuntu 16.04.

PS: I'm not used to catkin build and I have many packages build already with catkin_make and the transition doesn't seem easy, so I don't want to make any mess.

TSC21, 2017-12-11:
> Yeah I installed via "apt-get install geographiclib-tool". Is it the correct one? Because it's the only one I found. I'm on Ubuntu 16.04.

That installs the dataset updater scripts. You want `apt-get install libgeographic-dev`. Again, if you followed the instructions correctly, you wouldn't have to install it, cause rosdep would have done it for you.

tuloski, 2017-12-11:
ok, libgeographic-dev worked. I didn't use rosdep because I need to create my own plugins so I had to install from source. And the source installation instructions requires to use wstool and rosinstall_generator which I'd like to avoid.

TSC21, 2017-12-11:
What is the problem of that? You can use the install_generator and still write your own plugins and then build.

closed 2017-12-11 by TSC21