mavlink/mavros #885: FCU not connected in simulation

SarahMars, 2017-12-12:
### Issue details
I was able to simulate with this launch file but after I downgraded from PX4 firmware version 1.6.5 to 1.6.2, it stopped working.(1.6.5 wasn't working correctly for my actual quadrotor.)

Please let me know if any address like gcs_url is wrong.

```
<launch>
	<!-- vim: set ft=xml noet : -->
	<!-- example launch script for PX4 based FCU's -->

	<arg name="fcu_url" default="udp://:14540@127.0.0.1:14557" /> 
	<arg name="gcs_url" default="udp-b://127.0.0.1:14555@14550"/> 
<!--inside the "" was blank-->
	<arg name="tgt_system" default="1" />
	<arg name="tgt_component" default="1" />
	<arg name="log_output" default="screen" />

	<include file="$(find mavros)/launch/node.launch">
		<arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
		<arg name="config_yaml" value="$(find mavros)/launch/px4_config.yaml" />

		<arg name="fcu_url" value="$(arg fcu_url)" />
		<arg name="gcs_url" value="$(arg gcs_url)" />
		<arg name="tgt_system" value="$(arg tgt_system)" />
		<arg name="tgt_component" value="$(arg tgt_component)" />
		<arg name="log_output" value="$(arg log_output)" />
	</include>
</launch>

```

### MAVROS version and platform

Mavros: 0.20.0
ROS: Lunar
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[x] PX4 
Version: 1.6.2

### Diagnostics

```
    name: "mavros: FCU connection"
    message: "not connected"
    hardware_id: "udp://:14550@127.0.0.1:14557"

```

SarahMars, 2017-12-13:
It used to work with this and it doesn't work anymore
`roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"`

SarahMars, 2017-12-14:
Downgrading the firmware was making simulation not work. I erased and got the latest firmware again and then it worked.

closed 2017-12-14 by SarahMars

Seanmatthews, 2019-04-04:
Is there any further insight into this problem? I'm seeing this on master, with the most recent mavros from the apt repo.