mavlink/mavros #886: Offboard control not arming

SarahMars, 2017-12-12:
### Issue details
I followed this link [(https://dev.px4.io/en/ros/mavros_offboard.html)] and used exactly same code as here. It worked fine in the simulation, but with the actual copter it doesn't arm. It will keep showing me "Offboard enabled". In the simulation it's only displayed once and then moved onto "Vehicle armed". With an actual copter, eventually it will arm but it is switched to AUTO.LAND and OFFBOARD continuously.

### MAVROS version and platform

Mavros: 0.20.0
ROS: Lunar
Ubuntu: 16.04
### Autopilot type and version
[ ] ArduPilot
[x] PX4
Version: 1.6.2

Here is the result of `rosrun mavros offb_node`
```
[ INFO] [1513065237.705473527]: Offboard enabled
[ INFO] [1513065242.754505876]: Offboard enabled
[ INFO] [1513065247.804223358]: Offboard enabled
[ INFO] [1513065252.854399812]: Offboard enabled
[ INFO] [1513065257.905596186]: Offboard enabled
[ INFO] [1513065262.954086947]: Offboard enabled
[ INFO] [1513065278.009690786]: Offboard enabled
[ INFO] [1513065301.732780519]: Offboard enabled
[ INFO] [1513065320.177190688]: Offboard enabled
[ INFO] [1513065325.183543481]: Offboard enabled
[ INFO] [1513065330.230271946]: Offboard enabled
[ INFO] [1513065335.280053988]: Offboard enabled
```

Here is the result of ` rostopic echo /mavros/state`
```
header: 
  seq: 13
  stamp: 
    secs: 1513065217
    nsecs: 209551346
  frame_id: ''
connected: True
armed: False
guided: False
mode: "MANUAL"
system_status: 3
---
header: 
  seq: 14
  stamp: 
    secs: 1513065224
    nsecs: 121481688
  frame_id: ''
connected: True
armed: False
guided: False
mode: "MANUAL"
system_status: 3
---
header: 
  seq: 15
  stamp: 
    secs: 1513065226
    nsecs: 133732073
  frame_id: ''
connected: True
armed: False
guided: False
mode: "MANUAL"
system_status: 3
---
header: 
  seq: 16
  stamp: 
[... 477 lines cut ...]
    level: 2
    name: "mavros: Time Sync"
    message: "No events recorded."
    hardware_id: "/dev/ttyUSB0:57600"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "0"
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
        key: "Last dt (ms)"
        value: "0.000000"
      - 
        key: "Mean dt (ms)"
        value: "0.000000"
      - 
        key: "Last system time (s)"
        value: "0.000000000"
      - 
        key: "Time offset (s)"
        value: "0.000000000"
```
Thank you so much in advance.

vooon, 2017-12-12:
Is anything reported by mavros_node starting with `FCU:`?

SarahMars, 2017-12-12:
@vooon 
Here is the result of `rosrun mavros mavros_node`. I am using serial radio and its port is `ttyUSB0`. I am confused why it is saying `ttyACM0` even my launch file doesn't have it as URL.

```
[ INFO] [1513108931.584720710]: FCU URL: serial:///dev/ttyACM0
[ WARN] [1513108931.586334506]: init: message from ASLUAV, MSG-ID 201 ignored! Table has different entry.
[ WARN] [1513108931.586712807]: init: message from autoquad, MSG-ID 150 ignored! Table has different entry.
[ WARN] [1513108931.586762849]: init: message from autoquad, MSG-ID 152 ignored! Table has different entry.
[ WARN] [1513108931.587075951]: init: message from matrixpilot, MSG-ID 150 ignored! Table has different entry.
[ WARN] [1513108931.587121693]: init: message from matrixpilot, MSG-ID 151 ignored! Table has different entry.
[ WARN] [1513108931.587178502]: init: message from matrixpilot, MSG-ID 152 ignored! Table has different entry.
[ WARN] [1513108931.587232095]: init: message from matrixpilot, MSG-ID 153 ignored! Table has different entry.
[ WARN] [1513108931.587273967]: init: message from matrixpilot, MSG-ID 155 ignored! Table has different entry.
[ WARN] [1513108931.587316334]: init: message from matrixpilot, MSG-ID 156 ignored! Table has different entry.
[ WARN] [1513108931.587357114]: init: message from matrixpilot, MSG-ID 157 ignored! Table has different entry.
[ WARN] [1513108931.587398040]: init: message from matrixpilot, MSG-ID 158 ignored! Table has different entry.
[ WARN] [1513108931.587438627]: init: message from matrixpilot, MSG-ID 170 ignored! Table has different entry.
[ WARN] [1513108931.587478883]: init: message from matrixpilot, MSG-ID 171 ignored! Table has different entry.
[ WARN] [1513108931.587588075]: init: message from matrixpilot, MSG-ID 172 ignored! Table has different entry.
[ WARN] [1513108931.587627780]: init: message from matrixpilot, MSG-ID 173 ignored! Table has different entry.
[ WARN] [1513108931.587666573]: init: message from matrixpilot, MSG-ID 174 ignored! Table has different entry.
[ WARN] [1513108931.587704098]: init: message from matrixpilot, MSG-ID 175 ignored! Table has different entry.
[ WARN] [1513108931.587742469]: init: message from matrixpilot, MSG-ID 176 ignored! Table has different entry.
[ WARN] [1513108931.587780102]: init: message from matrixpilot, MSG-ID 177 ignored! Table has different entry.
[ WARN] [1513108931.587818099]: init: message from matrixpilot, MSG-ID 178 ignored! Table has different entry.
[ WARN] [1513108931.587855919]: init: message from matrixpilot, MSG-ID 179 ignored! Table has different entry.
[ WARN] [1513108931.587893599]: init: message from matrixpilot, MSG-ID 180 ignored! Table has different entry.
[ WARN] [1513108931.587931307]: init: message from matrixpilot, MSG-ID 181 ignored! Table has different entry.
[ WARN] [1513108931.587969356]: init: message from matrixpilot, MSG-ID 182 ignored! Table has different entry.
[ WARN] [1513108931.588007132]: init: message from matrixpilot, MSG-ID 183 ignored! Table has different entry.
[ WARN] [1513108931.588057912]: init: message from matrixpilot, MSG-ID 184 ignored! Table has different entry.
[ WARN] [1513108931.588110049]: init: message from matrixpilot, MSG-ID 185 ignored! Table has different entry.
[ WARN] [1513108931.588148198]: init: message from matrixpilot, MSG-ID 186 ignored! Table has different entry.
[ WARN] [1513108931.588451527]: init: message from paparazzi, MSG-ID 180 ignored! Table has different entry.
[ WARN] [1513108931.588492601]: init: message from paparazzi, MSG-ID 181 ignored! Table has different entry.
[ WARN] [1513108931.588544902]: init: message from paparazzi, MSG-ID 182 ignored! Table has different entry.
[ WARN] [1513108931.588596002]: init: message from paparazzi, MSG-ID 183 ignored! Table has different entry.
[ WARN] [1513108931.588633684]: init: message from paparazzi, MSG-ID 184 ignored! Table has different entry.
[ WARN] [1513108931.588929231]: init: message from slugs, MSG-ID 170 ignored! Table has different entry.
[ WARN] [1513108931.588968064]: init: message from slugs, MSG-ID 172 ignored! Table has different entry.
[ WARN] [1513108931.589006305]: init: message from slugs, MSG-ID 173 ignored! Table has different entry.
[ WARN] [1513108931.589044885]: init: message from slugs, MSG-ID 176 ignored! Table has different entry.
[ WARN] [1513108931.589083435]: init: message from slugs, MSG-ID 177 ignored! Table has different entry.
[ WARN] [1513108931.589120607]: init: message from slugs, MSG-ID 179 ignored! Table has different entry.
[ WARN] [1513108931.589157871]: init: message from slugs, MSG-ID 180 ignored! Table has different entry.
[ WARN] [1513108931.589195263]: init: message from slugs, MSG-ID 181 ignored! Table has different entry.
[ WARN] [1513108931.589237128]: init: message from slugs, MSG-ID 184 ignored! Table has different entry.
[ WARN] [1513108931.589275217]: init: message from slugs, MSG-ID 185 ignored! Table has different entry.
[ WARN] [1513108931.589313677]: init: message from slugs, MSG-ID 186 ignored! Table has different entry.
[ WARN] [1513108931.589351367]: init: message from slugs, MSG-ID 188 ignored! Table has different entry.
[ WARN] [1513108931.589395427]: init: message from slugs, MSG-ID 191 ignored! Table has different entry.
[ WARN] [1513108931.589433383]: init: message from slugs, MSG-ID 192 ignored! Table has different entry.
[ WARN] [1513108931.589470963]: init: message from slugs, MSG-ID 193 ignored! Table has different entry.
[ WARN] [1513108931.589508779]: init: message from slugs, MSG-ID 194 ignored! Table has different entry.
[ INFO] [1513108931.590336699]: serial0: device: /dev/ttyACM0 @ 57600 bps
[FATAL] [1513108931.623863652]: FCU: DeviceError:serial:open: No such file or directory
```

SarahMars, 2017-12-13:
@vooon 
Update, I tried offboard again and found out it's saying `FCU: Failsafe enabled: no RC and no offboard`. Even though I am like 3 meter away.

vooon, 2017-12-13:
`serial:///dev/ttyACM0` is default `~/fcu_url` parameter value. Check your launch, it should set node private params to change url.
Check your wiring, and that you complete configuration of FCU in GCS.

lamping7, 2017-12-20:
I don't think you fully understand what that code is doing. It is repeatedly checking the arm and mode and setting them if they are not offboard and armed. A failsafe is getting triggered, causing the mode to change to land (I'm assuming you have the settings under "Safety" in QGC set to land on data loss). This code then sees that and switches the mode back. This goes on in a cycle.

It looks like it gets enough setpoints, allows the switch and then enters failsafe from not getting enough setpoints and this happens repeatedly. I suspect connection issues and dropping messages.

What do you mean by "using serial radio"? How are you connected to the FCU and where is this code running?

SarahMars, 2017-12-26:
sorry for the late update, I found out it actually arms and tries to take off in the air but it seems like the flow sensor is not working correctly and won't stay in a position. Plus it is losing connection between quadrotor and the Linux laptop and switching off board control and AUTO.LAND frequently.

"using serial radio" means I am using ROS on my Linux laptop which is number one in this link. [hardware setup](https://dev.px4.io/en/ros/offboard_control.html). This code is running on my laptop.

The reason why I am using 1.6.2 is because I uploaded 1.6.5 several times but the quadrotor didn't take off. I downgraded to 1.6.2 and then it worked fine so I decided to keep using it.

closed 2017-12-26 by SarahMars