mavlink/mavros #887: Some MAVROS data not coming

TrishantRoy, 2017-12-13:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

TSC21, 2017-12-13:
how can you have APM and PX4 running on the same board?

```
[ INFO] [1513155449.759468635]: FCU: APM:Copter V3.5.4 (284349c3)
[ INFO] [1513155449.759729100]: FCU: PX4: 0384802e NuttX: 1bcae90b
```

TSC21, 2017-12-13:
Did you load the PX4 Firmware to the Pixhawk?

vooon, 2017-12-13:
@TSC21 APM runs on top of PX4 components, so this is APM:Copter 3.5.4.
@TrishantRoy please place logs inside code blocks. Did you setup stream rate parameters (use GCS)?

Deb version: `dpkg --status ros-indigo-mavros`

TrishantRoy, 2017-12-13:
@vooon I haven't set the stream parameters. Which of the parameters to set and to what value?

vooon, 2017-12-13:
First try: `rosrun mavros rate --all 10` http://wiki.ros.org/mavros#mavsys

TSC21, 2017-12-13:
> @TSC21 APM runs on top of PX4 components, so this is APM:Copter 3.5.4.

Ok I leave it to you then.

TrishantRoy, 2017-12-13:
`trishantroy@EurusHolmes:~$ rosrun mavros rate --all 10
[rosrun] Couldn't find executable named rate below /opt/ros/indigo/share/mavros`

khancyr, 2017-12-13:
rosrun mavros mavsys rate --all 10

TrishantRoy, 2017-12-13:
Okay... Still not happening... Only IMU data is coming

Is there a problem with baud rate?

TrishantRoy, 2017-12-13:
@vooon We can get data from PX4Flow as well, right?
That is the main data I want to get velocity.

TrishantRoy, 2017-12-26:
Can you confirm?
Coz I need to use the OFS to get data when connected to Pixhawk.

closed 2018-01-28 by TSC21