mavlink/mavros #890: waypoint: current_seq doesn't follow is_current right away

lamping7, 2017-12-14:
### Issue details

`/mavros/mission/waypoints` topic issue

`current_seq` doesn't follow `is_current = True` immediately

### MAVROS version and platform

Mavros: 0.21.5
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: 1.7.0-rc4

### Node logs

```
$ rostopic echo /mavros/mission/waypoints
---
current_seq: 0
waypoints: []
---
current_seq: 0
waypoints: 
  - 
    frame: 3
    command: 16
    is_current: True
    autocontinue: True
    param1: 0.0
    param2: 5.0
    param3: 0.0
    param4: nan
    x_lat: 39.1335868835
    y_long: -84.3651428223
    z_alt: 15.0
  - 
    frame: 3
    command: 16
    is_current: False
    autocontinue: True
    param1: 0.0
    param2: 5.0
    param3: 0.0
    param4: nan
    x_lat: 39.1337966919
    y_long: -84.3651809692
    z_alt: 15.0
---
current_seq: 1
waypoints: 
  - 
    frame: 3
    command: 16
    is_current: False
    autocontinue: True
    param1: 0.0
    param2: 5.0
    param3: 0.0
    param4: nan
    x_lat: 39.1335868835
    y_long: -84.3651428223
    z_alt: 15.0
  - 
    frame: 3
    command: 16
    is_current: True
    autocontinue: True
    param1: 0.0
    param2: 5.0
    param3: 0.0
    param4: nan
    x_lat: 39.1337966919
    y_long: -84.3651809692
    z_alt: 15.0
---
current_seq: 1                        <-----------------------------------------------------
waypoints: 
  - 
    frame: 3
    command: 16
    is_current: True                  <-----------------------------------------------------
    autocontinue: True
    param1: 0.0
    param2: 5.0
    param3: 0.0
    param4: nan
    x_lat: 39.1337966919
    y_long: -84.3651809692
    z_alt: 15.0
  - 
    frame: 3
    command: 16
    is_current: False
    autocontinue: True
    param1: 0.0
    param2: 5.0
    param3: 0.0
    param4: nan
    x_lat: 39.1338539124
    y_long: -84.3650360107
    z_alt: 15.0
---
current_seq: 1                        <-----------------------------------------------------
waypoints: 
  - 
    frame: 3
    command: 16
    is_current: True                  <-----------------------------------------------------
    autocontinue: True
    param1: 0.0
    param2: 5.0
    param3: 0.0
    param4: nan
    x_lat: 39.1338539124
    y_long: -84.3650360107
    z_alt: 15.0
  - 
    frame: 3
    command: 16
    is_current: False
    autocontinue: True
    param1: 0.0
    param2: 5.0
    param3: 0.0
    param4: nan
    x_lat: 39.1337203979
    y_long: -84.3647918701
    z_alt: 15.0
---
current_seq: 0
waypoints: 
  - 
    frame: 3
    command: 16
    is_current: True
    autocontinue: True
    param1: 0.0
    param2: 5.0
    param3: 0.0
    param4: nan
    x_lat: 39.1338539124
    y_long: -84.3650360107
    z_alt: 15.0
  - 
    frame: 3
    command: 16
    is_current: False
    autocontinue: True
    param1: 0.0
    param2: 5.0
    param3: 0.0
    param4: nan
    x_lat: 39.1337203979
    y_long: -84.3647918701
    z_alt: 15.0
---
```

vooon, 2017-12-14:
How you switch current wp when it reach end? Looks like problem in MISSION_CURRENT handler.

lamping7, 2017-12-14:
This appeared due to a unique case where the FCU mission is updated rapidly without reaching the waypoint. 

For example, if you have an algorithm running on an OBC that decides to send new mission waypoints to the FCU due to whatever, say it sees an obstacle or something, and it generates new waypoints before reaching that mission item. 

If you look just at the last two waypoint lists above, you'll see that the items don't change, but the current_seq is behind in the first, pointing to the incorrect index.

If this is a PX4 issue, I can take this conversation over there. Just let me know. Thanks.

vooon, 2017-12-14:
I don't tested in flight list changes (when you in AUTO.MISSION), can't say anything.
Anyway `current_seq` should update on MISSION_CURRENT coming from FCU.
Perhaps in this situation PX4 do not send it?

lamping7, 2017-12-14:
Perhaps. For me it is hard to tell if mavros doesn't update at first or if PX4 doesn't send it. I can check the mavros debug messages and set PX4 `mavlink verbose on` to check and see who is behind. This isn't in my immediate to-do list (it may be some time before I get to it), but I'll likely try to verify the messages before going over there with the issue.