mavlink/mavros #892: Wrong statements in code, output correct

ChristophTobler, 2017-12-14:
The function `quaternion_from_rpy()` (https://github.com/mavlink/mavros/blob/master/mavros/src/lib/ftf_quaternion_utils.cpp#L26) should actually be called `quaternion_from_ypr()` as the order is yaw, pitch and roll and not the other way around.

This is also why https://github.com/mavlink/mavros/blob/master/mavros/src/lib/ftf_frame_conversions.cpp#L30 is performing the correct transformation although the text/description is wrong. The description says first +180 degrees around X followed by +90 degrees around Z but it should be **-90** around Z. But since its not rpy but ypr +90 Z followed by +180 X is correct.

This would need some cleanup in several plugins I think. But before starting to change anything I would like someone to verify this.

TSC21, 2017-12-14:
> The function `quaternion_from_rpy()` (https://github.com/mavlink/mavros/blob/master/mavros/src/lib/ftf_quaternion_utils.cpp#L26) should actually be called `quaternion_from_ypr()` as the order is yaw, pitch and roll and not the other way around.

I understand your point, but what you actually feed into the function, as argument, is a `Eigen::Vector3d` which `XYZ` components are `RPY` respectively, and not `YPR`. What you do inside the function, then, is invert the argument inputs, which is also referenced in the comment https://github.com/mavlink/mavros/blob/master/mavros/src/lib/ftf_quaternion_utils.cpp#L28.

ChristophTobler, 2017-12-14:
@TSC21 Yes, that's why I said the output is correct. I can see that roll pitch and yaw arguments are in that order but the order also states which rotations will be performed first.
I think it's misleading if it's stated as rpy instead of ypr because if you just look at https://github.com/mavlink/mavros/blob/master/mavros/src/lib/ftf_frame_conversions.cpp#L30, you would think the calculation is wrong (because of the text and function call).

TSC21, 2017-12-14:
```c++
/**
 * @brief Static quaternion needed for rotating between ENU and NED frames
 * +PI/2 rotation about Z (Down) followed by +PI rotation around X (North) axis 
 * gives the ENU frame.  Similarly, a +PI/2 rotation about Z (Up) followed by
 * a +PI rotation about X (East) gives the NED frame.
 */
```
Does the above work to you? That way you actually do the rotation first on yaw, then on roll.

ChristophTobler, 2017-12-14:
@TSC21 Yes, that would work.

But I'm still skeptical towards `quaternion_from_rpy()`. Because if someone uses that function in a new plugin (or where ever) the assumption will be that roll will be performed first, then pitch and yaw which leads to an inconsistent result.

TSC21, 2017-12-14:
`quaternion_from_rpy()` naming comes from how the the argument is built, not how it is used. For avoiding trouble, that's why https://github.com/mavlink/mavros/blob/master/mavros/src/lib/ftf_quaternion_utils.cpp#L28 was added. The misleading texts probably are the ones in https://github.com/mavlink/mavros/blob/master/mavros/src/lib/ftf_frame_conversions.cpp, and that's why I suggested the above. Would you mind make the correction as above on a PR, since you referenced it? Thanks.

vooon, 2017-12-14:
Naming comes from bullet (`tf2::LinearMath` or tf::`), you can see that in test: https://github.com/mavlink/mavros/blob/master/mavros/test/test_quaternion_utils.cpp#L35

Also note that python `tf.transformations` function `quaternion_from_euler` do same.

We may rename to `quaternion_from_euler()`, but do that really needed? We have many places where use that function, so review may take long time.

TSC21, 2017-12-23:
Fixed in #895

closed 2017-12-23 by TSC21