mavlink/mavros #894: joule with px4 can't change to Offboard mode

raywang-nx, 2017-12-16:
Now i change pixhawk to px4, but i can't make it to change to offboard mode, neither using RC or mavros.
when i run mavsetp, it shows something like this:

rosrun mavros mavsetp local -v 0 0 0 0
Requesting OFFBOARD mode...
OFFBOARD mode are set.

and 

[ WARN] [1513421776.805711484]: CMD: Unexpected command 176, result 1

but pixhawk's red light blinks for maybe 4 times, with buzzer rings 3 times, and can't change it to offboard.

i think i need to send several position to it before, but everything seems the same.

and someone said that after the drone fly in the sky, change to alt-hold(may be other name i guess), then change to offboard and success, but i manage to change to position mode , but fail to change to offboard with the same situation like this.

### MAVROS version and platform

Mavros: ros-kinetic-mavros is already the newest version (0.21.5-0xenial-20171201-081620-0800).

ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version:  installed from latest qgc, maybe 1.7.0 i guess

### Node logs

joule@joule-570x-DVT3:~$ roslaunch mavros px4.launch 
... logging to /home/joule/.ros/log/97af8a86-e24f-11e7-ad78-f78ca70923e1/roslaunch-joule-570x-DVT3-7754.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://joule-570x-DVT3:36437/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 10.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
[... 386 lines cut ...]
    level: 0
    name: "mavros: Battery"
    message: "Normal"
    hardware_id: "/dev/ttyS1:921600"
    values: 
      - 
        key: "Voltage"
        value: "65.54"
      - 
        key: "Current"
        value: "-0.0"
      - 
        key: "Remaining"
        value: "-1.0"
---

### Check ID

joule@joule-570x-DVT3:~$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 5474 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 140, 141, 147, 24, 30, 31, 32, 36, 65, 70, 74, 76, 77, 87, 230, 231, 105, 111, 241, 245

TSC21, 2017-12-16:
`mavsetp` won't work the way you expect (it actually requires an update). Reade the offboard docs on the PX4 devguide and you will see you are missing something critical: you are not sending a stream of setpoints to the board before changing to Offboard mode. That'a why it does not change mode.

raywang-nx, 2017-12-18:
@TSC21 thanks a lot, now i understand what means setpoints, i am able to make it to the Offboard mode by publish /mavros/setpoint plugins, really appreciate!

closed 2017-12-18 by raywang-nx