mavlink/mavros #896: Error starting mavros

fnoop, 2017-12-18:
----

### Issue details

```
* /mavros/startup_px4_usb_quirkterminate called after throwing an instance of 'pluginlib::ClassLoaderException'
  what():  Unable to find package: mavros
================================================================================REQUIRED process [mavros-1] has died!
process has died [pid 5364, exit code -6, cmd /srv/maverick/software/ros/current/lib/mavros/mavros_node __name:=mavros __log:=/srv/maverick/var/log/ros/sitl/b33b5eb6-e316-11e7-9e7f-b827eb6d8aa4/mavros-1.log].
log file: /srv/maverick/var/log/ros/sitl/b33b5eb6-e316-11e7-9e7f-b827eb6d8aa4/mavros-1*.log
Initiating shutdown!
================================================================================
```

### MAVROS version and platform

Mavros: ? Current as installed from source instructions
ROS: Lunar
Ubuntu: Raspbian Stretch

### Autopilot type and version

[ x ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

See attached file for logs

[mavros-starfail.zip](https://github.com/mavlink/mavros/files/1569603/mavros-starfail.zip)

vooon, 2017-12-19:
Looks like problem is here:
```
 * /mavros/startup_px4_usb_quirkterminate called after throwing an instance of 'pluginlib::ClassLoaderException'
  what():  Unable to find package: mavros
```

Did you done `source /path/to/ws/dist/setup.bash`?

fnoop, 2017-12-19:
Yes, I'm using the same environment and startup manifests that I use for all platforms - they're written out with configuration management (puppet).
This is only since I added this new platform (raspbian stretch) and had to specify mavros build dependencies manually as the --deps doesn't work properly.

vooon, 2017-12-20:
Try to check where you call setup.bash. Did you install mavros in same place as roscore?
Also i assume that you build whole ros from sources.

Problem come from pluginlib, as i understand (but better look at code), it can't find `<mavros plugin="path/to/xml" />` tags in `package.xml`'s

fnoop, 2018-02-27:
This turned out to be lunar not working on raspbian stretch (debian 9).  Recompiled kinetic and works OK.