mavlink/mavros #899: Simulation in Gazebo + Mavros/SITL, error in Circle mode

monabf, 2017-12-19:
Hi all,

I have been simulating a drone in Gazebo using Mavros with SITL and ArduCopter, I've tried different modes and things, but I have a problem with the Circle mode. From the [ArduCopter documentation](http://ardupilot.org/copter/docs/circle-mode.html), I understand that when not overwriting yaw control the nose of the drone should always be pointing towards the center of the circle, which is not the case for me. The drone is circling fine, I have no error messages anywhere, but its nose is always pointing forward instead of towards the center of the circle. I am not overwriting any yaw commands from the autopilot so it looks like this could be a bug in the Gazebo + mavros/SITL setup. [Here](9https://gist.github.com/monabf/39ada6784216edc2d39fbf67cc450260) is my small script if you want to take a look. 

If anyone has ever tried using Circle mode with Gazebo I would be happy to know how it went! Any help is greatly appreciated!

### MAVROS version and platform

Mavros: 0.21.3
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[x] ArduPilot
[  ] PX4

### Check ID

```
rosrun mavros checkid
```

TSC21, 2017-12-23:
@monabf from my code analysis, this doesn't seem to have to do with MAVROS, but rrather something on the Gazebo sim and APM.

monabf, 2017-12-27:
Thanks for taking a look! I also asked on the Gazebo and Ardupilot forums, hopefully I'll get some answers there.

closed 2018-01-28 by TSC21

anish-pratheepkumar, 2019-12-09:
> Hi all,
> 
> I have been simulating a drone in Gazebo using Mavros with SITL and ArduCopter. Most modes work properly, but I have a problem with the Circle mode. From the [ArduCopter documentation](http://ardupilot.org/copter/docs/circle-mode.html), I understand that the nose of the drone should always be pointing towards the center of the circle, which is not the case for me. The drone is circling fine, I have no error messages anywhere, but its nose is always pointing forward instead of towards the center of the circle. I am not overwriting any yaw commands from the autopilot so it looks like this could be a bug in the Gazebo + mavros/SITL setup. [Here](https://gist.github.com/monabf/39ada6784216edc2d39fbf67cc450260) is my small script if you want to take a look.
> 
> If anyone has ever tried using Circle mode with Gazebo I would be happy to know how it went! Any help is greatly appreciated!
> 
> ### MAVROS version and platform
> Mavros: 0.21.3
> ROS: Kinetic
> Ubuntu: 16.04
> 
> ### Autopilot type and version
> [x] ArduPilot
> [ ] PX4