mavlink/mavros #900: Setting imu stream rate over udp

nav-go, 2017-12-20:
I'm using mavros to connect to a 3DR solo via udp using apm.launch

My current data rate at mavros/imu/data is only about 10 Hz. How do I increase this rate? 

rosservice call /mavros/set_stream_rate 0 100 1  does not do anything. Are higher rates possible over Wifi?

vooon, 2017-12-20:
Rate depends on APM firmware, maybe you can set higher rates via FCU parameters.
Ask at apm support forum.

nav-go, 2017-12-29:
Thanks for your guidance. I am able to increase IMU rate to 50Hz in Qgroundcontrol by changing a parameter (SR1_RAW_SENS) on the FCU. But the rate in mavros stays at 10Hz. What else can I do to make mavros access SR1 parameter. Where does mavros/imu/data get its data from?

vooon, 2017-12-30:
Check data_raw topic, it might be 50 Hz. On APM we have to combine several messages to make `data` message.
Data rate of `data` should be same as `ATTITUDE`, `data_raw` — `RAW_IMU`.

closed 2018-01-28 by TSC21