mavlink/mavros #905: Controlling velocity

SimonVleugels, 2017-12-23:
Hi,

I'm trying to control the velocity of a drone to change the position above a marker. I have been using `~setpoint_velocity/cmd_vel (geometry_msgs/TwistStamped) ` messages, but until now I have not been able to control the velocities. It looks like I don't even listens to what I ask.

I wrote some test code were the drone should hover when I switch to offboard mode. I do this by setting the velocity to 0 in each direction. I'm not using a GPS, so the drone should do it all by itself.

This is my test code where I send the velocity commands to the pixhawk.

```
    ros::Rate rate(50.0);
    this->vel_msg.twist.linear.x = this->xs;
    this->vel_msg.twist.linear.y = this->ys;
    this->vel_msg.twist.linear.z = this->zs;
    while(ros::ok() && !this->manual){
        this->vel_msg.header.stamp = ros::Time::now();
        this->vel_pub.publish(this->vel_msg);
        ros::spinOnce();
        rate.sleep();
    }
```

In [this](http://forum.erlerobotics.com/t/navigation-using-mavros/1000/2) forum post, they said I should use instead `/mavros/setpoint_raw/local mavros_msgs::PositionTarget` messages and set  the `PositionTarget::FRAME_BODY_OFFSET_NED` flag. But I have no clue how to begin with this.

TSC21, 2017-12-23:
@SimonVleugels you don't have to use `setpoint_raw`. Just update the `https://github.com/mavlink/mavros/blob/master/mavros/launch/px4_config.yaml#L111` to `BODY_NED`. PX4 does not support `BODY_OFFSET_NED` frame.

SimonVleugels, 2017-12-23:
Thank you for your comment. My px4_config.yaml is missing those lines. Can I just copy them, after that my drone will hover with a velocity of 0 in each direction without using a gps and using the `~setpoint_velocity/cmd_vel (geometry_msgs/TwistStamped)` message type?

```
# setpoint_position
setpoint_position:
  tf:
    listen: false           # enable tf listener (disable topic subscribers)
    frame_id: "map"
    child_frame_id: "target_position"
    rate_limit: 50.0

# setpoint_velocity
# None

# vfr_hud
# None

# waypoint
mission:
  pull_after_gcs: true  # update mission if gcs updates
```

SimonVleugels, 2017-12-23:
And I don't have to compile anything after changing the yaml file?

TSC21, 2017-12-23:
Are you using the deb package release or did you clone the repo to you catkin ws?

TSC21, 2017-12-23:
In any case, I recommend you to clone the latest master and build your ws.

SimonVleugels, 2017-12-23:
I took a clone from the repo a month ago. Darn that takes ages

SimonVleugels, 2017-12-23:
While building my ws again I got following error

```
==> Processing catkin package: 'libmavconn'
==> Building with env: '/opt/ros/kinetic/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/pi/ros_catkin_ws/build_isolated/libmavconn'
==> make -j4 -l4 in '/home/pi/ros_catkin_ws/build_isolated/libmavconn'
make[2]: *** No rule to make target '/home/pi/ros_catkin_ws/src/mavros/libmavconn/src/mavlink_helpers.cpp', needed by 'CMakeFiles/mavconn.dir/src/mavlink_helpers.cpp.o'.  Stop.
CMakeFiles/Makefile2:389: recipe for target 'CMakeFiles/mavconn.dir/all' failed
make[1]: *** [CMakeFiles/mavconn.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'libmavconn': 
  Command '['/opt/ros/kinetic/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2
```

TSC21, 2017-12-23:
`catkin clean --all -y` then `catkin build` again.

SimonVleugels, 2017-12-23:
`catkin_make clean --all -y ` gave following error. I'm following [this](http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi) (point 4) guide to install packages.

```
CMake Error at catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
```

SimonVleugels, 2017-12-23:
I don't know why its not working now, I haven been using this method always for adding extra packages

SimonVleugels, 2017-12-24:
@TSC21 Alright, I updated my ws and changed the `/mavros/launch/px4_config.yaml#L111` to `BODY_NED`. But still no "perfect" hover. 
I printed the message I'm sending out, and those values are the velocities I want, but when I listen to the `/mavros/setpoint_velocity/cmd_vel` I only get 0's 

Output send node: (xs,ys,zs are velocities in x,y,z direction)
```
xs : -0.0501582
ys : 0.0156759
zs : 0.0917009
1514120090.598762146
-----
xs : -0.0502775
ys : 0.0157128
zs : 0.0919202
1514120090.618715933
-----
xs : -0.0503969
ys : 0.0157498
zs : 0.0921394
1514120090.638766730
-----
xs : -0.0503969
ys : 0.0157498
zs : 0.0921394
1514120090.658698831
```

Output `rostopic echo /mavros/setpoint_velocity/cmd_vel` : 
```
---
header: 
  seq: 659
  stamp: 
    secs: 1514120090
    nsecs: 805063341
  frame_id: ''
twist: 
  linear: 
    x: 0.0
    y: 0.0
    z: 0.0
  angular: 
    x: 0.0
    y: 0.0
    z: 0.0
---
header: 
  seq: 660
  stamp: 
    secs: 1514120090
    nsecs: 818846327
  frame_id: ''
twist: 
  linear: 
    x: 0.0
    y: 0.0
    z: 0.0
  angular: 
    x: 0.0
    y: 0.0
    z: 0.0
---
header: 
  seq: 661
  stamp: 
    secs: 1514120090
    nsecs: 838844446
  frame_id: ''
twist: 
  linear: 
    x: 0.0
    y: 0.0
    z: 0.0
  angular: 
    x: 0.0
    y: 0.0
    z: 0.0
---
header: 
  seq: 662
  stamp: 
    secs: 1514120090
    nsecs: 858931466
  frame_id: ''
twist: 
  linear: 
    x: 0.0
    y: 0.0
    z: 0.0
  angular: 
    x: 0.0
    y: 0.0
    z: 0.0
---
```

TSC21, 2017-12-24:
Something in your node is wrong then. The output you are printing is not the output of your node, as if everything matches what I am seeing, your node is basically printing zeros, so probably it is being inited but not sending the values you want.

SimonVleugels, 2017-12-24:
I was forgetting the fact that while I'm switching back to manual mode, I send 0's. But after looking further in history of my terminal, the `/mavros/setpoint_velocity/cmd_vel` topic is giving the right velocities. But the drone still isn't doing what I want.

Output node:
```
header: 
  seq: 0
  stamp: 1514124012.539204947
  frame_id: 
twist: 
  linear: 
    x: 0.315529
    y: 0.242142
    z: 1
  angular: 
    x: 0
    y: 0
    z: 0

header: 
  seq: 0
  stamp: 1514124012.559152432
  frame_id: 
twist: 
  linear: 
    x: 0.316253
    y: 0.242735
    z: 1
  angular: 
    x: 0
    y: 0
    z: 0

header: 
  seq: 0
  stamp: 1514124012.579208919
  frame_id: 
twist: 
  linear: 
    x: 0.304509
    y: 0.345784
    z: 1
  angular: 
    x: 0
    y: 0
    z: 0

header: 
  seq: 0
  stamp: 1514124012.599125006
  frame_id: 
twist: 
  linear: 
    x: 0.307697
    y: 0.326133
    z: 1
  angular: 
    x: 0
    y: 0
    z: 0

[ INFO] [1514124012.619183203]: switchManual opvragen
[ INFO] [1514124012.619223487]: switchManual functie
[ INFO] [1514124012.750304789]: Offboard disabled

```

Output topic:
```
---
header: 
  seq: 374
  stamp: 
    secs: 1514124012
    nsecs: 539204947
  frame_id: ''
twist: 
  linear: 
    x: 0.315529108047
    y: 0.242141615599
    z: 1.0000000149
  angular: 
    x: 0.0
    y: 0.0
    z: 0.0
---
header: 
  seq: 375
  stamp: 
    secs: 1514124012
    nsecs: 559152432
  frame_id: ''
twist: 
  linear: 
    x: 0.316252857447
    y: 0.242735445499
    z: 1.0000000149
  angular: 
    x: 0.0
    y: 0.0
    z: 0.0
---
header: 
  seq: 376
  stamp: 
    secs: 1514124012
    nsecs: 579208919
  frame_id: ''
twist: 
  linear: 
    x: 0.304509475827
    y: 0.345784425735
    z: 1.0000000149
  angular: 
    x: 0.0
    y: 0.0
    z: 0.0
---
header: 
  seq: 377
  stamp: 
    secs: 1514124012
    nsecs: 599125006
  frame_id: ''
twist: 
  linear: 
    x: 0.307696796954
    y: 0.326132588089
    z: 1.0000000149
  angular: 
    x: 0.0
    y: 0.0
    z: 0.0
---
```

TSC21, 2017-12-24:
What do you mean with "not doing what you want"?

SimonVleugels, 2017-12-24:
Not moving with    
```
    x: 0.307696796954
    y: 0.326132588089
    z: 1.0000000149
```

But just random moving.

I want the drone to move itself 1m above an aruco marker. Via a camera (looking down) he gets the real time position (x,y,z distance) of the marker and with those distances combined with pid-controllors I get velocities in x,y,z direction so he would move itself.

Maybe a strange question but a positive value of x, is straight ahead? a positive value of y, is to the left? And a positive value of z is up?

TSC21, 2017-12-24:
The values you publish on your node should be in the ENU frame. MAVROS then converts them to the NED frame. But in this case you are working on the body frame, so I assume you are publishing the values on the with Forward-Left-Up frame. The aruco marker shouls be on the local frame, so I don't know why do you really want to send the velocities in the body frame, given you want the vehicle to adjust its position on the local frame.

SimonVleugels, 2017-12-24:
Isn't the marker also on the body frame? Because the detection of the marker gives a distance between the marker  and the center of an image in x,y,z-coördinates? For example my detection node detects the marker at a distance of x=1m, y=0.75m and z=1m from the center of my camera. And wil use these distances to fly 1m in x direction (of the drone) and 0.75m in y direction.
This way i only need the body frame, right? So if i want to publish in body frame, which topic should I use?

How I understand the body frame:
1. The origin is located at the center of the camera of the drone.
2. The X-axis points forward.
3. The Y-axis is the right side of drone.
4. The Z-axis upward.

SimonVleugels, 2017-12-27:
@TSC21 Or should I go with the /rc/override topic? And hope that will work? Sorry for spamming but I have not much time anymore

SimonVleugels, 2017-12-27:
@TSC21 I'm a bit confused with the body and local frame types, I can't find much about it or I'm searching wrong. 

I want to be able to say to the drone, fly forward with a velocity of 0.15 m/s for example. And forward is just forward, like how a car drives, not to the north. But like you said the values need to be published in East North Up frame, so how can I do it the way I want?

TSC21, 2017-12-27:
The marker is in the camera frame. So you should add a transform from the camera frame to the `base_link` frame (ROS body frame). Should be as simple as a doing a translation.

closed 2018-01-28 by TSC21

bobolodrone, 2020-11-26:
hi everyone, i have a similar problem. i publish velocity on setpoint_velocity/cmd vel but nothing happened. i  checked the pwm of the motors with rc/out but it didn't change. how i can solve it?
this is what i pblished 
 seq: 329
  stamp:
    secs: 0
    nsecs:         0
  frame_id: ''
twist:
  linear:
    x: 5.0
    y: 0.0
    z: 0.0
  angular:
    x: 0.0
    y: 0.0
    z: 0.0

dylantzx, 2021-07-01:
Hello, I am currently facing the same problem. Has a solution been found for this issue yet? @SimonVleugels @TSC21 @bobolodrone 

I intend to use velocity setpoints to control my drone and am publishing them to the `/mavros/setpoint_velocity/cmd_vel` topic. Similarly, I want my drone to move **forward** (in the direction that it is currently facing) by say 1m/s if I require it to. 

I am sending these setpoints for example (where yaw is just a TwistStamped msg):
```
yaw.header.stamp.sec = ros::Time::now().toSec();
yaw.header.stamp.nsec = ros::Time::now().toNSec();
yaw.header.frame_id = "base_link";
yaw.twist.linear.x = 1;
yaw.twist.linear.y = 0;
local_yaw_pub.publish(yaw);
```

Here is an example of my output:
```
header: 
  seq: 2315
  stamp: 
    secs: 8831
    nsecs: 791239424
  frame_id: "base_link"
twist: 
  linear: 
    x: 1.0
    y: 0.0
    z: 0.0
  angular: 
    x: 0.0
    y: 0.0
    z: 0.0

```
However, I am my drone is currently moving **North** and not forward. May I know if I am doing this wrong or if there is a workaround for this?

dylantzx, 2021-07-01:
> @SimonVleugels you don't have to use `setpoint_raw`. Just update the `https://github.com/mavlink/mavros/blob/master/mavros/launch/px4_config.yaml#L111` to `BODY_NED`. PX4 does not support `BODY_OFFSET_NED` frame.

Oh actually, I just tried changing the mav_frame in px4_config.yaml to BODY_NED and I think I got it to work. I'll do more testing to see if it affects anything else. Thank you!