mavlink/mavros #906: simple script using mavros setpoint_position

ericjohnson97, 2017-12-24:
I'm new to mavros. I am trying to use the simple takeoff script as seen [here on the px4 docs](https://dev.px4.io/en/ros/mavros_offboard.html) work with the APM flight stack. I changed the "OFFBOARD" mode to the APM "GUIDED" mode, but every time I run the script the drone arms, but doesn't takeoff. I'm testing with STIL and gazebo. I can't seem to figure out what I'm doing wrong. Thanks in advance!

here is my code
```

/**
 * @file nedTest.cpp
 */

#include <fstream>

#include <ros/ros.h>
#include <mavros_msgs/CommandBool.h>
#include <geometry_msgs/PoseStamped.h>
#include <mavros_msgs/SetMode.h>
#include <mavros_msgs/State.h>

mavros_msgs::State current_state;
void state_cb(const mavros_msgs::State::ConstPtr& msg){
    current_state = *msg;
}

int main(int argc, char **argv)
{
    ros::init(argc, argv, "offb_node");
    ros::NodeHandle nh;

    ros::Subscriber state_sub = nh.subscribe<mavros_msgs::State>("mavros/state", 1, state_cb);
    ros::Publisher local_pos_pub = nh.advertise<geometry_msgs::PoseStamped>("mavros/setpoint_position/local", 10);
    ros::ServiceClient arming_client = nh.serviceClient<mavros_msgs::CommandBool>("mavros/cmd/arming");
    ros::ServiceClient set_mode_client = nh.serviceClient<mavros_msgs::SetMode>("mavros/set_mode");

    //the setpoint publishing rate MUST be faster than 2Hz
    ros::Rate rate(20.0);

    // wait for FCU connection
    while(ros::ok() && current_state.connected){
        ros::spinOnce();
        rate.sleep();
    }

    geometry_msgs::PoseStamped pose;
    pose.pose.position.x = 0;
    pose.pose.position.y = 0;
    pose.pose.position.z = 2;

    //send a few setpoints before starting
    for(int i = 100; ros::ok() && i > 0; --i){
	local_pos_pub.publish(pose);
        ros::spinOnce();
        rate.sleep();
    }

    mavros_msgs::SetMode offb_set_mode;
    offb_set_mode.request.custom_mode = "GUIDED";

    mavros_msgs::CommandBool arm_cmd;
    arm_cmd.request.value = true;

    ros::Time last_request = ros::Time::now();

    while(ros::ok()){
      //std::cout << pose << std::endl;
        if( current_state.mode != "GUIDED" && (ros::Time::now() - last_request > ros::Duration(5.0))){
            if( set_mode_client.call(offb_set_mode)){
                ROS_INFO("Offboard enabled");
            }
            last_request = ros::Time::now();
        } else {
            if( !current_state.armed && current_state.mode == "GUIDED" && (ros::Time::now() - last_request > ros::Duration(5.0))){
[... 496 lines cut ...]
    level: 0
    name: "mavros: Battery"
    message: "Normal"
    hardware_id: "udp://127.0.0.1:14551@14555"
    values: 
      - 
        key: "Voltage"
        value: "12.59"
      - 
        key: "Current"
        value: "0.0"
      - 
        key: "Remaining"
        value: "93.0"
---

```

### Check ID
OK. I got messages from 1:1.

---
Received 1643 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 132, 65, 136, 152, 147, 150, 24, 27, 29, 30, 32, 33, 163, 36, 165, 42, 173, 178, 182, 62, 193, 74, 35, 164, 100, 241, 116, 125

TSC21, 2017-12-26:
@khancyr maybe you can lend a hand here?

TSC21, 2018-01-28:
@ericjohnson97 did you find the solution yet?

ericjohnson97, 2018-01-29:
So here is what my team and I have learned. We noticed in a previous issue that you cannot issue setpoint position commands when you are on the ground, even though in the px4 tutorial they can get it to takeoff with setpoint position(i assume a px4 flight stack difference). Once we had the drone takeoff we were able to have it move to the destinations we desired. Some things that we have noticed about this command is that the drone sometimes moves very slowly like the FCU is being overwhelmed by messages. We think this because if we break our script the drone will complete the maneuver at a more reasonable pace. We were also interested in using setpoint_raw command so we can control the speed at which the drone moves, but with the limited testing we have done so far the drone seems to continue past the desired position. Is this command supposed to work the same as its setpoint position counter part?

TSC21, 2018-05-04:
> We were also interested in using setpoint_raw command so we can control the speed at which the drone moves, but with the limited testing we have done so far the drone seems to continue past the desired position. Is this command supposed to work the same as its setpoint position counter part?

This is all a matter of how you are generating the commands and how the autopilot controller handles them. Closing this now.

closed 2018-05-04 by TSC21

yuchnw, 2019-02-01:
> So here is what my team and I have learned. We noticed in a previous issue that you cannot issue setpoint position commands when you are on the ground, even though in the px4 tutorial they can get [...]

I have the same issue as you mentioned in this post. Could you please tell me how you managed to solve this issue? Thanks!

yuchnw, 2019-02-05:
@ericjohnson97 Thank you for the reply! Right now I can make the quad takeoff to the sky, however, when I publish to the `/mavros/setpoint_position/local` topic, the quad doesn't change its path. It just keeps flying upward.

Here is the code I have (I'm writing in Python).
```
def setLocation():
    locPub = rospy.Publisher('/mavros/setpoint_position/local',PoseStamped,queue_size=10)
    loop_rate = rospy.Rate(10)
        
    while not rospy.is_shutdown():
        goalPos = PoseStamped()
        goalPos.pose.position.x = 0
        goalPos.pose.position.y = 0
        goalPos.pose.position.z = 15
        goalPos.header.stamp = rospy.Time.now()
        goalPos.header.frame_id = 'base_link'

        locPub.publish(goalPos)
        loop_rate.sleep()
```
So basically after the quad successfully takeoff, I just call the `setLocation()` function. But it doesn't work. Do you have any idea what might cause this issue? Thanks!