mavlink/mavros #909: Moving drone using setpoint_raw/target_local

SimonVleugels, 2017-12-27:
Hi,

I'm planning to use the `setpoint_raw/target_local` topic, but have some questions about the use of it.
The idea is that my drone will move in x,y directions to a certain location. I have the distances between the drone and the destination, which get updated 30 times a second. I have no GPS so I was planning to use velocities to move the drone.

1. Is FRAME_BODY_OFFSET_NED  the coordinate frame I need (z needs to be up)
    -> setting this flag: `moveMsg.coordinate_frame = mavros_msgs::PositionTarget::FRAME_BODY_OFFSET_NED;`

2. In the documentation I found the type_mask field. Am I supposed to set all the ignore flags? (IGNORE_P*, IGNORE_AF*, IGNORE_YAW, IGNORE_YAW_RATE)
Like this: 
```
moveMsg.type_mask = mavros_msgs::PositionTarget::IGNORE_AFX |
                    mavros_msgs::PositionTarget::IGNORE_AFY |
                    mavros_msgs::PositionTarget::IGNORE_AFZ ;
```

3. When I use only the velocity part of the message, am I able to move the drone?
Like this:

```
moveMsg.velocity.x = 0.1;
moveMsg.velocity.y = 0.1;
moveMsg.velocity.z = 0.0;
```

I'm trying to get my drone moving for weeks now, but I cant get the hang of it.
Thanks a head!

TSC21, 2017-12-27:
PX4 or APM?

SimonVleugels, 2017-12-27:
Px4

TSC21, 2017-12-27:
> Px4

So you will have to use `PositionTarget::FRAME_LOCAL_NED` or `PositionTarget::FRAME_BODY_NED`.

SimonVleugels, 2017-12-27:
But what is the difference? I'm a bit confused what the best is. I have distances in forward, left way, not north east.

SimonVleugels, 2017-12-27:
I'm really appreciating your help, but have you also answers on the other questions? I'm in a really short time space

SimonVleugels, 2017-12-27:
Oke thanks I will investigate it. I have been searching for some information about coordinate types, but couldn't find much about it. But the thing I want to do is possible without GPS?

TSC21, 2017-12-27:
> But the thing I want to do is possible without GPS?

Yes.

SimonVleugels, 2017-12-27:
Thank you! Have you also an answer on my two other questions?

TSC21, 2017-12-27:
> 2. In the documentation I found the type_mask field. Am I supposed to set all the ignore flags?
> (IGNORE_P*, IGNORE_AF*, IGNORE_YAW, IGNORE_YAW_RATE)
> Like this:
> ```
> moveMsg.type_mask = [...]

Yes.

> 3. When I use only the velocity part of the message, am I able to move the drone?
> Like this:
> 
> ```
> moveMsg.velocity.x = 0.1;
> moveMsg.velocity.y = 0.1;
> moveMsg.velocity.z = 0.0;
> ```

Depends if you are setting things correctly on your publishing node, but it should work - question: why not using `setpoint_velocity` plugin? You wouldn't have to set the bitmask.

SimonVleugels, 2017-12-27:
I have tried to use it, but haven't been able to get it working.

So all the flags except those needs to be set?
```
uint16 IGNORE_AFY = 128
uint16 IGNORE_AFZ = 256
uint16 FORCE = 512
```

SimonVleugels, 2017-12-28:
I have been playing around, but it looks like my drone is ignoring me. I have been using `FRAME_BODY_OFFSET_NED` and for testing purposes sending consecutively -0.4 m/s in a direction and the other directions 0. Have tested all the direction. The drone seems to move always to the right, maybe because the wind is almost every time coming from the left. Is 0.4 m/s not fast enough?

The message I send:
```
header: 
  seq: 0
  stamp: 1514467976.462257475
  frame_id: drone
coordinate_frame: 9
type_mask: 4039
position: 
  x: 0
  y: 0
  z: 0
velocity: 
  x: 0
  y: -0.4
  z: 0
acceleration_or_force: 
  x: 0
  y: 0
  z: 0
yaw: 0
yaw_rate: 0

```

After `rostopic echo /mavros/setpoint_raw/local`

```
---
header: 
  seq: 613
  stamp: 
    secs: 1514467976
    nsecs: 462257475
  frame_id: drone
coordinate_frame: 9
type_mask: 4039
position: 
  x: 0.0
  y: 0.0
  z: 0.0
velocity: 
  x: 0.0
  y: -0.4
  z: 0.0
acceleration_or_force: 
  x: 0.0
  y: 0.0
  z: 0.0
yaw: 0.0
yaw_rate: 0.0
---
```

My code:

```
void DroneControl::initRos(ros::NodeHandle n)
{
    ...
    this->velocity_pub = n.advertise<mavros_msgs::PositionTarget>("mavros/setpoint_raw/local", 10);
    ...
}

void DroneControl::dronebesturing()
{
    ROS_INFO("in dronebesturing");
    ros::Rate rate(50.0);

    while(ros::ok() && !this->manual){
        mavros_msgs::PositionTarget velocity_msg;
        velocity_msg.coordinate_frame = mavros_msgs::PositionTarget::FRAME_BODY_OFFSET_NED;
        velocity_msg.header.frame_id = "drone";
        velocity_msg.type_mask = mavros_msgs::PositionTarget::IGNORE_PX |
                                 mavros_msgs::PositionTarget::IGNORE_PY |
                                 mavros_msgs::PositionTarget::IGNORE_PZ |
                                 mavros_msgs::PositionTarget::IGNORE_AFX |
                                 mavros_msgs::PositionTarget::IGNORE_AFY |
                                 mavros_msgs::PositionTarget::IGNORE_AFZ |
                                 mavros_msgs::PositionTarget::FORCE |
                                 mavros_msgs::PositionTarget::IGNORE_YAW |
                                 mavros_msgs::PositionTarget::IGNORE_YAW_RATE ;
        velocity_msg.header.stamp = ros::Time::now();

        velocity_msg.velocity.x = this->xs;
        velocity_msg.velocity.y = this->ys;
        velocity_msg.velocity.z = this->zs;

        if(this->testen){
            velocity_msg.velocity.x = 0.0;
            velocity_msg.velocity.y = -0.4;
            velocity_msg.velocity.z = 0.0;
        }

        cout << velocity_msg << endl;
        this->velocity_pub.publish(velocity_msg);
        ros::spinOnce();
        rate.sleep();
    }
    ROS_INFO("switchManual opvragen");
    switchManual();
    return;
}
```

TSC21, 2017-12-28:
You continue to use a frame of reference that is not supported by PX4 even after I told you that you have to use BODY_NED or LOCAL_NED (no OFFSET). But that's your call

SimonVleugels, 2017-12-28:
Ow I didn't meant to do that. After changing it to BODY_NED, it seems that the drone doesn't something. Now when sending 0.2 in any direction (the others 0.0) the drone shoots in to the air. Switching to LOCAL_NED does the same.

Updated code

```
void DroneControl::dronebesturing()
{
    ROS_INFO("in dronebesturing");
    ros::Rate rate(50.0);

    while(ros::ok() && !this->manual){
        mavros_msgs::PositionTarget velocity_msg;
        velocity_msg.coordinate_frame = mavros_msgs::PositionTarget::FRAME_BODY_NED;
        velocity_msg.header.frame_id = "drone";
        velocity_msg.type_mask = mavros_msgs::PositionTarget::IGNORE_PX |
                                 mavros_msgs::PositionTarget::IGNORE_PY |
                                 mavros_msgs::PositionTarget::IGNORE_PZ |
                                 mavros_msgs::PositionTarget::IGNORE_AFX |
                                 mavros_msgs::PositionTarget::IGNORE_AFY |
                                 mavros_msgs::PositionTarget::IGNORE_AFZ |
                                 mavros_msgs::PositionTarget::FORCE |
                                 mavros_msgs::PositionTarget::IGNORE_YAW |
                                 mavros_msgs::PositionTarget::IGNORE_YAW_RATE ;
        velocity_msg.header.stamp = ros::Time::now();

        velocity_msg.velocity.x = this->xs;
        velocity_msg.velocity.y = this->ys;
        velocity_msg.velocity.z = this->zs;

        if(this->testen){
            velocity_msg.velocity.x = 0.0;
            velocity_msg.velocity.y = -0.4;
            velocity_msg.velocity.z = 0.0;
        }

        cout << velocity_msg << endl;
        this->velocity_pub.publish(velocity_msg);
        ros::spinOnce();
        rate.sleep();
    }
    ROS_INFO("switchManual opvragen");
    switchManual();
    return;
}
```

SimonVleugels, 2017-12-30:
Anyone has an idea?

padmashree9110, 2018-01-02:
@SimonVleugels 
Hi,
 I tried testing the above code and it is working as expected.
For below settings,

- (vx, vy, vz): (0.0,-0.4,0.0) => drone is moving towards South (-ve Y-axis)

- (vx,vy,vz): (2.0,0.0,0.0) => drone is moving towards East (+ve X-axis)

Did you find the similar observation?

Regards
Padma

SimonVleugels, 2018-01-02:
@padmashree9110 
Thank you for confirming the code is working. Unfortunate when I'm testing it, the drone will start flying up and wont stop. Can you maybe supply me with the node you used for testing my code? So if your code would work, I know there is something wrong with the rest of my code.

Kind regards,
Simon

padmashree9110, 2018-01-03:
@SimonVleugels 
You can refer this code
https://gist.github.com/anonymous/b9ff7b1a24c41104073d2b56917351a1

SimonVleugels, 2018-01-18:
@padmashree9110 thank you for the code. So far I have been able to successfully test my code in gazebo. In real life I have tested without props to safely switch to offboard mode and back, this was a success. When I tested my code outside (I asked for 0.4m/s in x direction), he flew backwards instead of to the right. This caused an accident, 2 props destroyed, 1 damaged.

I feel a bit annoyed because in simulation it works perfect but in real life the drone is living his own life.

My full node:
https://gist.github.com/SimonVleugels/78ccccacb204e1262ad46967b401cf12

closed 2018-01-28 by TSC21