mavlink/mavros #913: test_mavros/launch/px4/iris_empty_world_offboard_ctl.launch issue

FaboNo, 2018-01-04:
I am trying to use the test file you have created to run with px4 and a model from the package Rotors (from ethz-asl) 
Actually I created a watkin_ws including Rotors, Firmware (px4), mavros and mavlink
Then catkin build without errors

then from catkin_ws: 
roslaunch src/mavros/test_mavros/launch/px4/iris_empty_world_offboard_ctl.launch

it appears (see below) that the some executables (manual_input, commander, mc_mixer, attitude_estimator, mc_att_control and mavlink) are not created during the catkin build process.

However, after checking the Firmware CMakeList I do not see how to generate them, do you have any suggestion or do I have to post on px4/Firmware?

Thank you very much for you advice

Fabrice
----

### Issue details

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
redefining global property: pi
when processing file: /home/fabrice/catkin_demo_ws/src/rotors_simulator/rotors_description/urdf/iris_base.xacro
xacro.py is deprecated; please use xacro instead
started roslaunch server http://fabrice-N650DU:45133/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/
 * /sitl_test_offboard_ctl/

PARAMETERS
 * /iris/MPP_XY_FF: 0.0
 * /iris/MPP_XY_P: 1.0
 * /iris/MPP_XY_VEL_D: 0.01
 * /iris/MPP_XY_VEL_I: 0.0
 * /iris/MPP_XY_VEL_MAX: 2.0
 * /iris/MPP_XY_VEL_P: 0.01
 * /iris/MPP_Z_P: 2.0
 * /iris/MPP_Z_VEL_P: 0.3
 * /iris/mavlink/mavlink_fcu_url: udp://localhost:1...
 * /iris/mixer: i
 * /iris/robot_description: <?xml version="1....
 * /iris/tf_prefix: 
 * /iris/vehicle_model: iris
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
[... 290 lines cut ...]
[  ] ArduPilot
[ x] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics

WARNING: no messages received and simulated time is active.
Is /clock being published?

### Check ID

```
rosrun mavros checkid
```

TSC21, 2018-01-04:
@FaboNo this package is not updated for ages now. I'm planning to update it to use with `PX4/sitl_gazebo` but I'm currently swamped with other stuff.

FaboNo, 2018-01-06:
@TSC21 oh I see, I will try to do it on my side and if I succeed I will post it; in case other people want to do the same think :)

FaboNo, 2018-05-05:
@TSC21 Yes I wrote a program in C++ and it works. Thank you for your concern.
I am trying to do the same thing with the offboard mode with our drone and I will probably let you know if I face some issues. Wish you a good week-end

TSC21, 2018-05-05:
Oh ok, I thought you were planning to update `test_mavros` to use `sitl_gazebo` instead. Thanks anyway. A good weekend for you too.

closed 2018-05-05 by TSC21