mavlink/mavros #915: [/mavros/local_position/local] does not appear to be published yet - with px4 mavros_posix_sitl

FaboNo, 2018-01-06:
### Issue details
I implemented the example provided on https://dev.px4.io/en/ros/mavros_offboard.html 
I launched roslaunch px4 mavros_posix_sitl.launch on one terminal and my code in another one.
It works fine but then I added a subscriber to mavros/local_postion/local
ros::Subscriber sub_ = nh.subscribe("/mavros/local_position/local", 10, reached_cb);
in order to be able to to send several setpoints and check when one setpoint has been reached or not.
but i noticed that nothing is published by mavros on that topic.

rostopic hz mavros/local_position/local
WARNING: topic [/mavros/local_position/local] does not appear to be published yet
 
however /mavros/global_position/local appears to be published by mavros.

Do you have any suggestion?

Thank you very much

Fabrice
### MAVROS version and platform

Mavros: ?0.18.4
ROS: ?Kinetic
Ubuntu: ?16.04

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

Version: ?3.7.1?

### Node logs
copy output of mavros_node. Usually console where you run roslaunch

roslaunch px4 mavros_posix_sitl.launch
... logging to /home/fabrice/.ros/log/96089c72-f31b-11e7-abcb-b0359f2c094d/roslaunch-fabrice-N650DU-1909.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://fabrice-N650DU:34997/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
[... 281 lines cut ...]
        key: Timesyncs since startup
        value: 2
      - 
        key: Frequency (Hz)
        value: 6.666667
      - 
        key: Last dt (ms)
        value: -9.797200
      - 
        key: Mean dt (ms)
        value: -7.485100
      - 
        key: Last system time (s)
        value: 73.971628000
      - 
        key: Time offset (s)
        value: 5.924453120
---
### Check ID

rosrun mavros checkidReceived 5442 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 140, 141, 147, 24, 30, 31, 32, 33, 74, 76, 85, 87, 102, 230, 105, 111, 241, 242, 245

```

FaboNo, 2018-01-06:
But it appears that:
/mavros/local_position/odom  is published and provides the local position with covariance (actually nav_msgs/Odometry)
so I can work with it:)

TSC21, 2018-01-06:
Is somewhat hard to understand what you want here. One thing is what you described in the first place, the other is the `odom` plugin, which has nothing to do with the first issue. `odom` provides local position with covariance **to** the FCU but it is used on the **estimators**, not in the controllers.

TSC21, 2018-01-06:
What's your PX4 Fimrware version, `sitl_gazebo` version and MAVROS version? (git commit tags please)

FaboNo, 2018-01-06:
@TSC21 oh yes good to know that odom is used by the estimator and not the controllers. I was echoing the different /mavros/local_position and as I saw this one was published ...
Regarding the version (git tag) 
mavros 0.21.2
Firmware (Including px4, sitl gazebo) v1.7.2 
Hope I answer your question correctly

danividanivi, 2018-01-09:
Why do you expect position to be published in mavros/local_position/local?
I don't see that topic in the [wiki](http://wiki.ros.org/mavros#Subscribed_Topics) neither it is in my rostopic list.
Are you sure the information you want is not in mavros/local_position/pose or mavros/global_position/local? The second one is the topic I use to check if my aircraft arrived to the point indicated.

closed 2018-01-28 by TSC21

yuchnw, 2019-02-10:
@FaboNo Hi. I am having the same issue as you described here. Did you solve that problem? Thanks!

CMcColl1658, 2019-08-04:
I'm hving similar issues with the current position
  ros::Subscriber currentPos = nh.subscribe<geometry_msgs::PoseStamped>("/uav2/mavros2/global_position/local", 10, pose_cb);

The topic is publishing the data fine and can be seen using rostopic echo however the topic does not say it has any subscribers even when i the node running.

yuchnw, 2019-08-09:
Hey forgot to post my updates. Have you set the **custom_mode** to **OFFBOARD**? I did that and it solved my issue. Think maybe you can give it a try.

yuchnw, 2019-08-09:
And make sure to send the **OFFBOARD** command continuously. For some reason if I only define it at the start it won't last long. So I called this command every time after I updated the position info.

clancky1658, 2019-08-09:
I’m using apm and that didn’t help.... any other ideas?