mavlink/mavros #916: How to test my serial interface

wazzan88, 2018-01-08:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details
Hi ,
I am a beginner in field Ros and mavlink
I hav a serial connection between FPGA Zedboard and my raspberry Pi3
I installed Ros and mavros on the Rsy pi and I sent am Hallo world messages from FPGA using Uart interface to Rsy pi I could see the messages using Terminal linux using minicom programm on the ACM0 interface in Rsy pi
so 
how could test the libmavconn to see if mavros could see this message
please if there is any examples to mak my work easier or details  
thank u very much

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: ?0.18.4? 
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

vooon, 2018-01-08:
I'm not quite understand what you trying to do. To test libmavconn enable debug output via rqt (or by rosconsole.config), key: `log4j.logger.ros.rosconsole_bridge.mavcon`.

Starting from 0.18 we use template generator to include all dialects, so editing of `mvlink_dialect.h` is not needed any more.

Better give difinitions.xml file more specific name, usually `<autopilot>.xml`.

wazzan88, 2018-01-11:
I am following the instruction to install mavros on raspberry pi 3
# 1. Create the workspace: unneded if you already has workspace
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws
catkin init
wstool init src

# 2. Install MAVLink
#    we use the Kinetic reference for all ROS distros as it's not distro-specific and up to date
rosinstall_generator --rosdistro kinetic mavlink | tee /tmp/mavros.rosinstall

# 3. Install MAVROS: get source (upstream - released)
rosinstall_generator --upstream mavros | tee -a /tmp/mavros.rosinstall

# 4. Create workspace & deps
wstool merge -t src /tmp/mavros.rosinstall
wstool update -t src -j4
rosdep install --from-paths src --ignore-src -y

# 5. Install GeographicLib datasets:
./src/mavros/mavros/scripts/install_geographiclib_datasets.sh

# 6. Build source
catkin build
when I excute the number 6 it takes long time then I becom error because of the memory

so I installed useing `apt-get` for installation:
sudo apt-get install ros-kinetic-mavros ros-kinetic-mavros-extras
Then install GeographicLib datasets by running the `install_geographiclib_datasets.sh` script:
 wget https://raw.githubusercontent.com/mavlink/mavros/master/mavros/scripts/install_geographiclib_datasets.sh 
 ./install_geographiclib_datasets.sh
but when I go inside the installed files they are not the same as the first method I found inside mavros 
cmake  launch  mavros_plugins.xml  package.xml
and not the whole code 
my question what is the defrennce and also if I want the code to make changes on it ?
what should i do should I do the first method but what aboutvirtual memory problem ?

vooon, 2018-01-13:
2 hours is small, usually rpi much slower (one of c++ weakness). But you can setup distcc and use USB hard drive to improve IO speed.

Normally build system will rebuild only changed files, so it should be faster.

wazzan88, 2018-01-14:
thank u very much for ur reply :)
now I want to work on the FPGA side 
I hav zynq as processor so I want just to use Heartbeat message 
my goal to send it from my Zynq to Mavros
at the beginnng I generated just minimal.xml
and I had the files I added them to my project
so I am working on sdk it belongs to Vivado
so I  should use mavlink_msg_heartbeat_pack function to pack the message and mavlink_msg_to_send_buffer to make it in serial way to send it
but I think I should write functions to send it using uart or ?? 
 because I hav not found functions to do that ??
I should write MAVLINK_SEND_UART_BYTES() function to do these task ??
if what I said right ????
----------------------------
my second question is
could I use your code that u made in mavros ??
Msgbuffer.h , interface.h , serial.h+cpp
to do this task but without Ros u think it will works if I copied the Code??
--------------------------
Third question
I will suppose that I did the previous step and I wrote the Code that will send my mavlink messages one of them Tapas message from FPGA
and in Mavros there is no such message .
where should I adjust mavros ??
I know that I should 1. add the xml file of the new messages into mavlink
2. maybe I should add this new message into mavros_msgs/msg or ??
are there any parts that I should add this message to ??
----------------
actually I read ur Code it is really perfect thank u for ur effort
for my project. I think I can not use the parts that belongs to APM and PX4
or the mavros_extras what do u think ??
are there any parts that I do not need ??
-----------
thank u very much

TSC21, 2018-01-28:
@wazzan88 if you have other questions please open a new issue which subject represents your question. Closing this now

closed 2018-01-28 by TSC21