mavlink/mavros #917: How to use fake_gps plugin?

lajoiepy, 2018-01-09:
### Issue details

Hi!
I am looking into using the fake_gps plugin in mavros_extras to fly indoor. I have access to vision data in the form of a transform between frames called vision_origin and vision_pose_estimate.
Here's my config:
```
# fake_gps
fake_gps:
  # select data source
  use_mocap: false         # ~mocap/pose
  mocap_transform: false   # ~mocap/tf instead of pose
  use_vision: false       # ~vision (pose)
  # origin (default: Zürich)
  geo_origin:
    lat: 47.3667          # latitude [degrees]
    lon: 8.5500           # longitude [degrees]
    alt: 408.0            # altitude (height over the WGS-84 ellipsoid) [meters]
  eph: 2.0
  epv: 2.0
  satellites_visible: 5   # virtual number of visible satellites
  fix_type: 3             # type of GPS fix (default: 3D)
  tf:
    listen: true
    send: false           # send TF?
    frame_id: "vision_origin"       # TF frame_id
    child_frame_id: "vision_pose_estimate" # TF child_frame_id
    rate_limit: 10.0      # TF rate
  gps_rate: 5.0           # GPS data publishing rate
```
The plugin seems to work fine, but I am a little confused about the output I should expect. Nothing is published on /mavros/global_position/local nor on the other global_position topics.
Note that I am using PX4 with LPE and that I have set the `MAV_USEHILGPS` to `enabled` and `LPE_FUSION` to `1` (only GPS).

### MAVROS version and platform

Mavros: 0.21.2
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

PX4
Version: 1.6.3

TrishantRoy, 2018-01-18:
Even I have the doubt.... Will post here if I am able to solve it.
@lajoiepy  Do post here if you get the solution.

@TSC21  @ChristophTobler  Can you help?

Thanks.

TSC21, 2018-01-18:
Did try setting `LPE_FAKE_ORIGIN` PX4 parameter to 1?

TrishantRoy, 2018-01-18:
Parameter not found. I may be using APM. But it should depend on that, right?

What is position lock?

TSC21, 2018-01-18:
> Parameter not found. I may be using APM. But it shouldn't depend on that, right?

LPE is a PX4 position estimator. So I think you have the answer there.

> What is position lock?

It's when the autopilot actually knows where it is.

TSC21, 2018-01-18:
> Did try setting `LPE_FAKE_ORIGIN` PX4 parameter to 1?
>
> **Edit:** actually this should not be required, as this is published om uorb as vehicle_gps_position. Do you get a position lock at all?

With the above I was actually answering to @lajoiepy question.

TrishantRoy, 2018-01-18:
Can't fake_gps be used for APM?

Found an implementation in APM similar to fake_gps.
https://github.com/quadrotor-IITKgp/emulate_GPS/blob/master/mavros_vision/src/vision.cpp

I would be very thankful if you could tell if I could use it.

For adding a plugin, we have to make changes in CMakeList and mavros plugins file, right?

Thanks

TSC21, 2018-01-18:
`fake_gps` should work with APM. No need for adding any new plugin.

TrishantRoy, 2018-01-18:
Thanks @TSC21 

How to use and test fake_gps?

Thanks

TSC21, 2018-01-18:
I am not an APM user so I can't actually help you with that.

TSC21, 2018-01-18:
@khancyr maybe you can give a hand here?

TrishantRoy, 2018-01-19:
@TSC21  How to use and test fake_gps on PX4?

If required, I will use the PX4 firmware.

TSC21, 2018-01-20:
@TrishantRoy first of all, you need to be sure of what flight stack you want to use, which you definitely aren't. Both flight stacks have similarities but are also different in many ways, starting with the features that each have available. After that, without knowing what is the purpose of the usage for `fake_gps`, is kinda difficult to recommend how to use it, 'cause I don't even know what type of position source you want to convert to fake GPS coordinates. It can be a transform that you listen to, it can be a vision topic, it can be a mocap topic. If you have one of this sources, then on `px4_config.yaml`, you need to set the respective parameter. Assuming you want to use this on your place of testing, you also need to add the corresponding geographic origin (the LLA of where you are, instead of the Zurich's geographic coordinates).
If you are using PX4, then you also need to set `MAV_USEHILGPS` parameter on PX4 side, so the autopilot is able to use the fake GPS coordinates you are sending through MAVLink `HIL_GPS` msgs (which is what `fake_gps` plugin does). For APM, supposedly, this should work out-of-the-box without having to set a parameter for using `HIL_GPS` msgs as source of GPS data.
The above information is more than enough for using this plugin, so I'm closing this issue.

closed 2018-01-20 by TSC21