mavlink/mavros #918: Getting external compass magnetic field data through mavros

8Pedro8, 2018-01-09:
Hi,
I'm using using cuav autopilot pixhawk (ver 2.4) with px4 firmware on it (uploaded with Qgroundcontrol) and i have cuav neo m8n gps which has a compass and is able to be connected to pixhawk. And i'm just able to use px4 firmware, because of imu output rate, i was able to get high imu rate (200 hz) with this firmware (just this firmware) after some modification.

first, is it possible to do this, getting external compass data with mavros ? and how can i do it ?

I have tried to disable pixhawk's main compass in Qgroundcontrol with this param:
CAL_MAG0_EN -> disable
but that did not help me. 

ROS: Indigo
Ubuntu: 14.04

vooon, 2018-01-10:
I don't know any special message for that, but for IMU message FCU should send magn selected for navigation work.
By default it should use external one.

8Pedro8, 2018-01-10:
Hi vooon, thanks for the quick reply,
today i have tried Ardupilot flight stack, it was as you said and worked, but i'm not able to get more than 50 hz with it.

what i have done to get higher frequency : 
  for PX4 firmware : 
    I have added this two lines in /opt/extras.txt (created the folder in pixhawks sdcard):
      mavlink stream -d /dev/ttyACM0 -s ATTITUDE -r 200
      mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 200
    and with this i'm able to get 200 hz output on PX4 firmware.

  for Ardupilot firmware :
    that approach, didn't work, and i'm just able to increase it to 50 hz with:
       rosservice call /mavros/set_stream_rate 0 200 1

any suggest on how i can increase the imu rate on Ardupilot firmware or getting external compass data with PX4 firmware ?

vooon, 2018-01-10:
Perhaps APM have strict limit... And new per-message interface is not yet supported.
But i'm pretty sure that in PX4 magn operates similar, ask for support at http://discuss.px4.io/ .

closed 2018-05-10 by TSC21