mavlink/mavros #922: PX4 unable to control attitude

OHussain, 2018-01-19:
### Issue details

Hello,
I am using the PX4 SITL in MAVROS for offboard attitude control. I started with the sample position control code from the user guide, and modified it to hold its position with no problem. Now I am trying to do the same with attitude control and it appears to be rejecting the offboard mode. This is peculiar as I have a flag in my code to verify that the offboard control is enabled, but the Pixhawk flashes the red rejection light when attempting to switch to offboard. I suspect it has something to do with the publishing, but am unsure. Any help would be greatly appreciated.

### Code

#include <ros/ros.h>
#include <geometry_msgs/PoseStamped.h>
#include <geometry_msgs/Quaternion.h>
#include <sensor_msgs/Imu.h>
#include <geometry_msgs/Vector3.h>
#include <std_msgs/Float64.h>
#include <mavros_msgs/CommandBool.h>
#include <mavros_msgs/SetMode.h>
#include <mavros_msgs/State.h>
#include <mavros_msgs/RCIn.h>
#include <tf/transform_listener.h>
#include <tf/transform_datatypes.h>
#define Pi 3.14159

geometry_msgs::Vector3 Current_euler;
mavros_msgs::State current_state;
mavros_msgs::RCIn current_RC;

double Time = 0;

void state_cb(const mavros_msgs::State::ConstPtr& msg){
    current_state = *msg;
}

void RC_cb(const mavros_msgs::RCIn::ConstPtr& msg){
    current_RC = *msg;
}
void imuqua_cb(const sensor_msgs::Imu msg){
	tf::Quaternion IMU_q(msg.orientation.x,msg.orientation.y,msg.orientation.z,msg.orientation.w);
	tf::Matrix3x3 m(IMU_q);
	m.getRPY(Current_euler.x,Current_euler.y,Current_euler.z);
	Current_euler.z=-Current_euler.z;
	Current_euler.y=-Current_euler.y;
}
	bool wasArmed = false; //Check value for arming

int main(int argc, char **argv)
{
    ros::init(argc, argv, "offb_node");
    ros::NodeHandle nh;

    ros::Subscriber state_sub = nh.subscribe<mavros_msgs::State>
            ("mavros/state", 10, state_cb);
	ros::Subscriber RC_sub = nh.subscribe<mavros_msgs::RCIn>
            ("mavros/rc/in", 10, RC_cb);
	ros::Subscriber imuqua_sub = nh.subscribe<sensor_msgs::Imu>
            ("mavros/imu/data", 10, imuqua_cb);

 	ros::Publisher Eu_pub = nh.advertise<geometry_msgs::Vector3>//attitutude_pub
             ("Current_Euler", 10); 
	ros::Publisher Des_pub = nh.advertise<geometry_msgs::Vector3>//attitutude_pub
             ("Desired_Euler", 10); 
	 ros::ServiceClient arming_client = nh.serviceClient<mavros_msgs::CommandBool>
            ("mavros/cmd/arming");
   	 ros::ServiceClient set_mode_client = nh.serviceClient<mavros_msgs::SetMode>
            ("mavros/set_mode");
	
	ros::Publisher pub_att = nh.advertise<geometry_msgs::PoseStamped>("/mavros/setpoint_attitude/attitude",100);
	ros::Publisher pub_thr = nh.advertise<std_msgs::Float64>("mavros/setpoint_attitude/att_throttle",100);
	geometry_msgs::PoseStamped cmd_att;
	std_msgs::Float64 cmd_thr;

  int count = 1;
[... 550 lines cut ...]
      - 
        key: Frequency (Hz)
        value: 10.000422
      - 
        key: Last dt (ms)
        value: -0.069912
      - 
        key: Mean dt (ms)
        value: 0.010035
      - 
        key: Last system time (s)
        value: 264.676425000
      - 
        key: Time offset (s)
        value: 1516319345.540253162
---

### Check ID

OK. I got messages from 1:1.

---
Received 7514 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 140, 141, 147, 30, 31, 32, 36, 65, 70, 74, 76, 77, 83, 230, 231, 105, 111, 241, 245

OHussain, 2018-05-04:
No. We used setpoint_attitude/raw imstead of setpoint_attitude/attitude to work around the problem, but the original issue is still there. I’ll close it, but I have to wonder what setpoint_attitude/attitude is meant for.

closed 2018-05-04 by OHussain

TSC21, 2018-05-04:
> I’ll close it, but I have to wonder what setpoint_attitude/attitude is meant for.

Did you use it properly? Did you generate the attitude and the thrust setpoints and published them to the right topics? What MAVROS version were you even using? From my analysis you are using an old MAVROS version where the `setpoint_attitude` didn't even have the `message_filters` to sync the the attitude and thrust topics (you are using `Float64` for the Thrust topic when the current msg is `mavros_msgs/Thurst`.

TSC21, 2018-05-04:
This is the current code for a reference: https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_attitude.cpp

viswanarayanans, 2021-02-20:
I have a similar problem. I did "rostopic list". But "setpoint_attitude/attitude" is not in the list. I could see only "setpoint_attitude/cmd_vel" and "setpoint_attitude/thrust". I am using mavros with PX4. Is there anything that I need to specify to see the topic?