mavlink/mavros #923: problem with /mavros/setpoint_velocity

OliviaNocentini, 2018-01-19:
If I use ` /mavros/setpoint_raw/local ` to send the desired velocity to the drone, I read from ` /mavros/local_position/velocity` the same velocity.
Instead if I use ` /mavros/setpoint_velocity/cmd_vel ` to send the desired velocity to the drone in 
`/mavros/local_position/velocity` I read a velocity that is not the one that I published.Anyone had this problem??
What is the correct topic that I have to use to send a desired velocity to the vehicle?

ashwinvk94, 2018-04-06:
```/mavros/setpoint_raw/local``` is not used to set the desired velocity.

I am also trying to understand which topic should be used to send the desired velocity to the drone.

Can someone explain which of the below topics should be used and the differences between each:
```/mavros/setpoint_velocity/cmd_vel```
```/mavros/setpoint_velocity/cmd_vel_unstamped```
```/mavros/setpoint_attitude/cmd_vel```

TSC21, 2018-05-04:
> Instead if I use ` /mavros/setpoint_velocity/cmd_vel ` to send the desired velocity to the drone in 
`/mavros/local_position/velocity` I read a velocity that is not the one that I published. Anyone had this problem??

That depends on the `MAV_FRAME` you are using.

> ```/mavros/setpoint_velocity/cmd_vel```
```/mavros/setpoint_velocity/cmd_vel_unstamped```
```/mavros/setpoint_attitude/cmd_vel```
```/mavros/setpoint_raw/local```

First two are for controlling the velocity on local or body frames, with the only difference being on the Twist ROS msg (one has a timestamp, the other doesn't). The 3rd is for controlling body angular rates. the last allows you to do any of the other three but you have to manually set the bitmask.

closed 2018-05-04 by TSC21

Mettikodeva, 2023-06-21:
> 
Sorry for opening this issue again.
where and  when should I set the MAV_FRAME?
should I set it in launch file or in code?

Mettikodeva, 2023-06-21:
> That depends on the `MAV_FRAME` you are using.
>
>
> First two are for controlling the velocity on local or body frames, with the only difference being on the Twist ROS msg (one has a timestamp, the [...]

@TSC21 

sorry for tagging you, I have several questions.
- where and when should I set the MAV_FRAME?
- should I set it in launch file or in code?
- is BODY_NED deprecated?
I've tried to set the MAV_FRAME with
```
rosparam set /mavros/setpoint_velocity/mav_frame BODY_NED
```
but it seems doesn't work. I run a command to move on x only, but it moves on x, y, and z