mavlink/mavros #926: error roslaunch mavros

lp02781, 2018-01-24:
i had installed mavros on ubuntu 16.04 ROS distro kinetic, but when i connect my pixhawk to laptop and running `roslaunch mavros apm.launch`, this error appears 
```

```

lp02781@Lenovo:~$ roslaunch mavros apm.launch
... logging to /home/lp02781/.ros/log/24d26d28-00ca-11e8-baff-acb57df86549/roslaunch-Lenovo-9655.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Lenovo:44840/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_url: /dev/ttyACM0:57600
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
 * /mavros/global_position/tf/frame_id: map
 * /mavros/global_position/tf/global_frame_id: earth
 * /mavros/global_position/tf/send: False
 * /mavros/global_position/use_relative_alt: True
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: base_link
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: map
 * /mavros/local_position/tf/child_frame_id: base_link
 * /mavros/local_position/tf/frame_id: map
[... 104 lines cut ...]
[ WARN] [1516772884.179505232]: init: message from slugs, MSG-ID 184 ignored! Table has different entry.
[ WARN] [1516772884.179520417]: init: message from slugs, MSG-ID 185 ignored! Table has different entry.
[ WARN] [1516772884.179535454]: init: message from slugs, MSG-ID 186 ignored! Table has different entry.
[ WARN] [1516772884.179551037]: init: message from slugs, MSG-ID 188 ignored! Table has different entry.
[ WARN] [1516772884.179567204]: init: message from slugs, MSG-ID 191 ignored! Table has different entry.
[ WARN] [1516772884.179582009]: init: message from slugs, MSG-ID 192 ignored! Table has different entry.
[ WARN] [1516772884.179596853]: init: message from slugs, MSG-ID 193 ignored! Table has different entry.
[ WARN] [1516772884.179611900]: init: message from slugs, MSG-ID 194 ignored! Table has different entry.
[ INFO] [1516772884.179991857]: serial0: device: /dev/ttyACM0 @ 57600 bps
[FATAL] [1516772884.180150691]: FCU: DeviceError:serial:open: No such file or directory
================================================================================REQUIRED process [mavros-2] has died!
process has finished cleanly
log file: /home/lp02781/.ros/log/24d26d28-00ca-11e8-baff-acb57df86549/mavros-2*.log
Initiating shutdown!
================================================================================
[mavros-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

```

lp02781, 2018-01-24:
what should i do??

vooon, 2018-01-25:
You should provide correct URL to FCU.
```
[FATAL] [1516772884.180150691]: FCU: DeviceError:serial:open: No such file or directory
```

closed 2018-01-28 by TSC21