mavlink/mavros #927: Stream on serial sometimes doesn't start

tuloski, 2018-01-24:
If I use a serial connection with the autopilot (PX4 v2 with latest arducopter) I have to relaunch mavros several time to see the stream. I don't even receive the heartbeat, then suddenly it works after a few resets of the node. I also tried to send an heartbeat without success.

If I use mavros via UDP (mavros on a pc which communicates to a GS) I never have any problem, but the ground station has to send an heartbeat to start the stream.

Is there something that has to syncronize with the autopilot to start the stream? Also set stream rate doesn't help. Is the heartbeat not arriving?

vooon, 2018-01-25:
First: check wiring. Need to be sure that problem not related to physical connection problem. Check with multimeter Rx-Tx, Tx-Rx, GND-GND.

Then check what programs may use serial tty's. For example ModemManager.

tuloski, 2018-02-02:
The physical connection is ok, I checked it, and as I wrote, after a while it starts, and when it starts it never stops. Then if I stop mavros I have to repeat again and try 5-10 times before starting again. Is there a timeout to receive the first heartbeat?

vooon, 2018-02-02:
No, mavros will wait first heartbeat forever. Timeout only starts when first HB received, and resets on each HB message.
It used to detect when autopilot stop talking.

tuloski, 2018-02-15:
And who is starting the communication? Is mavros requesting a stream to the autopilot? Might it be that the stream request is not arriving (maybe the serial drops some bytes and the message is lost)?

vooon, 2018-02-15:
Did you configure firmware to run mavlink on used TELEM port?

tuloski, 2018-02-15:
Yeah, and it's working sometimes. I'll try to check what is mavros receiving via serial.
Another question: is the "ground station" (mavros) answering with an heartbeat every time it receives an heartbeat from the autopilot?

vooon, 2018-02-15:
Mavros always try to send HEARTBEAT to FCU ([if param `conn/heartbeat_rate` != 0][1]).
Message data: [timer callback][2].

[1]: https://github.com/mavlink/mavros/blob/master/mavros/launch/apm_config.yaml#L10
[2]: https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/sys_status.cpp#L820

tuloski, 2018-02-15:
Ok, found the issue.

Long story short: I'm an idiot. 
Long story: between the mavros device and the autopilot I have an arduino which just reads from the serial and resubmit to the other serial (both ways). When mavros starts and open the serial (USB), the arduino resets and sometimes it freezes (I'll check why but it's weird) closing the connection. I'll try to disable the DTR and RTS on the serial opening on mavros so the arduino won't reset.

Sorry for the loss of time :(

closed 2018-02-15 by vooon