mavlink/mavros #928: set_stream_rate not updating in mavros V0.22.0

soutieslak, 2018-01-25:
I recently moved from ROS Lunar back to Kinetic but made a backup of all my packages before the change.

I created a new catkin workspace and cloned the latest mavros version, 0.22.0. I am making use of the /mavros/imu/data topic for my project and need a fast stream rate. So I always change the stream rate as follows: rosservice call /mavros/set_stream_rate 0 100 1

I noticed that it was not updating the steam rate at all, so then I reloaded my backed up package of mavros, which is version 0.21.5 and performed exactly the same command:
rosservice call /mavros/set_stream_rate 0 100 1

And this updated the steam rate.

Is this a bug/issue in the new version 0.22.0 of mavros?

Maybe it is related to this oaks question:
https://github.com/mavlink/mavros/issues/900

vooon, 2018-01-26:
There is no any changes in set_stream_rate for a while. But in recent commit we start to use Mavlink 2.0 as a default. Try to set `fcu_protocol: v1.0`.

TSC21, 2018-05-04:
@soutieslak still an issue?

soutieslak, 2018-05-08:
@TSC21 I cannot comment yet. My project is on pause, but I continued to just use version 0.21.5 together with the "rosservice call /mavros/set_stream_rate 0 100 1" to get the correct steam rate I needed. I the project resumes then I will test it again and give feedback.

TSC21, 2018-05-08:
@soutieslak try setting the mavlink to protocol to v1.0

TSC21, 2018-05-31:
Closing this for now. Please reopen if you are still facing this issue.

closed 2018-05-31 by TSC21