mavlink/mavros #929: Can't catkin build Mavros on Odroid XU4

syc7446, 2018-01-25:
Hello, I have a question about doing catkin build Mavros on Odroid XU4.
I have installed Ubuntu 16.04 and Ros Kinetic on Odroid XU4.
And in order to get Mavros working, I followed instructions from here: [https://github.com/mavlink/mavros/blob/master/mavros/README.md#source-installation](url), which is the Mavros source installation.

However, I got the following error message when I did catkin build.

```
odroid@odroid:~/catkin_ws$ catkin build
-----------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/kinetic
Workspace:                   /home/odroid/catkin_ws
-----------------------------------------------------------
Source Space:       [exists] /home/odroid/catkin_ws/src
Log Space:          [exists] /home/odroid/catkin_ws/logs
Build Space:        [exists] /home/odroid/catkin_ws/build
Devel Space:        [exists] /home/odroid/catkin_ws/devel
Install Space:      [unused] /home/odroid/catkin_ws/install
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------
[build] Found '6' packages in 0.0 seconds.                                                                                                                                              
[build] Package table is up to date.                                                                                                                                                    
Starting  >>> mavlink                                                                                                                                                                   
Starting  >>> mavros_msgs                                                                                                                                                               
Finished  <<< mavlink                      [ 6.2 seconds ]                                                                                                                              
Starting  >>> libmavconn                                                                                                                                                                
Finished  <<< mavros_msgs                  [ 7.3 seconds ]                                                                                                                              
Finished  <<< libmavconn                   [ 1.1 seconds ]                                                                                                                              
Starting  >>> mavros                                                                                                                                                                    
________________________________________________________________________________________________________________________________________________________________________________________
Errors     << mavros:make /home/odroid/catkin_ws/logs/mavros/build.make.002.log                                                                                                         
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/gcs_bridge.dir/src/gcs_bridge.cpp.o] Error 4
make[1]: *** [CMakeFiles/gcs_bridge.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/mavros_node.dir/src/mavros_node.cpp.o] Error 4
make[1]: *** [CMakeFiles/mavros_node.dir/all] Error 2
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/param.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/waypoint.cpp.o] Error 4
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/sys_status.cpp.o] Error 4
make[1]: *** [CMakeFiles/mavros_plugins.dir/all] Error 2
make: *** [all] Error 2
cd /home/odroid/catkin_ws/build/mavros; catkin build --get-env mavros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
........................................................................................................................................................................................
Failed     << mavros:make                  [ Exited with code 2 ]                                                                                                                       
Failed    <<< mavros                       [ 5 minutes and 3.5 seconds ]                                                                                                                
Abandoned <<< mavros_extras                [ Unrelated job failed ]                                                                                                                     
Abandoned <<< test_mavros                  [ Unrelated job failed ]                                                                                                                     
[build] Summary: 3 of 6 packages succeeded.                                                                                                                                             
[build]   Ignored:   None.                                                                                                                                                              
[build]   Warnings:  None.                                                                                                                                                              
[build]   Abandoned: 2 packages were abandoned.                                                                                                                                         
[build]   Failed:    1 packages failed.                                                                                                                                                 
[build] Runtime: 5 minutes and 11.9 seconds total.
```

I believe that's the swap issue so I also set up the swap partition as follows:
```
odroid@odroid:~/catkin_ws$ free
              total        used        free      shared  buff/cache   available
Mem:        2038452      331832     1576964         964      129656     1669423
Swap:       1023996      215044      808952
```

I've been stuck here for awhile. Any ideas or suggestions? Thanks in advance!

vooon, 2018-01-26:
You don't have enough RAM to run 4 compillers in parallel. Limit by `-j2`.

syc7446, 2018-01-26:
Thanks @vooon ! `-j2` worked perfectly.

closed 2018-01-26 by syc7446