mavlink/mavros #933: local_position data issue

aaraujo11, 2018-01-30:
### Issue details

**1)** I connected the Jetson TX1 through serial port between /dev/ttyTHS1 and TelemetryPort2 of PixHawk Cube 2.1, with a 921600 baudrate.
I launched the mavros (apparently without any issues), then set the stream data rate to 10Hz `rosservice call /mavros/set_stream_rate "stream_id: 0 message_rate: 10 on_off: true"` or also `rosrun mavros mavsys rate --all 10`, i start to receive data at approx. 10Hz to some topics without any issue, but then when i check the topics of `/mavros/local_position/pose ... odom.... velocity` i dont have any data (even when i arm and flight). 
So to be sure that i was receiving the specific [mavlink ID 32 LOCAL_POSITION_NED](http://mavlink.org/messages/common#LOCAL_POSITION_NED) msg, i run the `rosrun mavros checkid` and i verify that there wasn't any 32 ID. Should i need to activate in the some flag on the params list in the pixhawk side?

**2)** Its is possible to set permanently the stream data rate? Because everytime i reboot the pixhawk i need to set again the stream rate to start receive data.

**3)**  Why im receiving the warning msgs at the beginning of launch `[ WARN] [1517338283.510153420]: init: message from ASLUAV, MSG-ID 201 ignored! Table has different entry.` in the mavros version 0.17.3 for example didn't appear.

**4)** Time to time i receive `HP: requesting home position` msg why? I need to define a home position, does the pixhawk set one home position automatically when armed?

Thank you in advance. 

### MAVROS version and platform

Mavros: 0.22.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[X] ArduPilot
[  ] PX4

Version: Copter 3.5.4

### Node logs

```
started roslaunch server http://tegra-ubuntu:59792/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/fcu_url: /dev/ttyTHS1:921600
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: True
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: base_link
 * /mavros/local_position/tf/frame_id: world
 * /mavros/local_position/tf/send: True
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
 * /mavros/plugin_blacklist: ['actuator_contro...
 * /mavros/plugin_whitelist: []
 * /mavros/px4flow/frame_id: px4flow
 * /mavros/px4flow/ranger_fov: 0.0
 * /mavros/px4flow/ranger_max_range: 5.0
 * /mavros/px4flow/ranger_min_range: 0.3
[... 349 lines cut ...]
    values: 
      - 
        key: Voltage
        value: 13.01
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 99.0
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 3782 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 129, 2, 1, 136, 137, 24, 147, 150, 152, 27, 29, 30, 193, 33, 35, 36, 165, 42, 178, 182, 62, 65, 74, 77, 163, 241, 116, 125

```

TSC21, 2018-02-01:
1. I don't know how it works with APM, but the logical is that you can only receive the local position from the vehicle if know where it is (position lock). So the question is, are you feeding any kind of position source to the vehicle?
2. APM internals can probably let you set the stream on the vehicle starting config, so you don't have to use MAVROS for it.
3. That will soon be fixed by https://github.com/mavlink/mavros/issues/834. Though, not a problem.
4. That just confirms what I said in 1. - no position lock, no home position set. That's basically warning you that you don't have home position set, which the autopilot should set automatically when it gets a position lock (it does not set when arms, but rather when it actually is able to estimate is position).

aaraujo11, 2018-02-02:
Hi @TSC21, 
You are right, the place where i was testing, somehow didn't have enough data/signal to perform lock position, i change the place of test, and that's it, the controller starts to stream local position. Thank you very much!

closed 2018-02-02 by aaraujo11

TSC21, 2018-02-02:
Boa sorte!

aaraujo11, 2018-02-02:
Obrigado Nuno!