mavlink/mavros #936: Implement Safety Feature

TrishantRoy, 2018-02-02:
Hey,
I have using mavros for Off-board control (Pixhawk, Quadcopter).

I am using the /mavros/rc/override topic to publish the channel values.

But, as a safety feature, I want to regain manual control using the transmitter on a particular condition, say, on using one of the unused channels.

I had read that publishing the value 65535 on the channel will lead to regain in manual mode.
(Is this correct?)

The code:
`msg_override.channels[0] = 65535;
 msg_override.channels[1] = 65535;
 msg_override.channels[2] = 65535;
 msg_override.channels[3] = 65535;
 rc_override_pub.publish(msg_override);
ros::shutdown();`

This isn't working.

What is the mistake? Is there any other alternative to implement the safety feature?

Thanks.

TSC21, 2018-02-02:
Not MAVROS related. Bring this to the firmware repositories.

closed 2018-02-02 by TSC21

TrishantRoy, 2018-02-02:
Just found a previous issue in which it said to publish 0 to regain control.
https://github.com/mavlink/mavros/issues/22

[image]

I will try this and post my result here.

Thanks.

TrishantRoy, 2018-02-02:
Hey,
You have to publish 0 to regain manual control.

`msg_override.channels[0] = 0;
        msg_override.channels[1] = 0;
        msg_override.channels[2] = 0;
        msg_override.channels[3] = 0;
        rc_override_pub.publish(msg_override);
        ros::shutdown();`