mavlink/mavros #937: Bug: terminate called after throwing an instance of 'std::bad_weak_ptr'

TSC21, 2018-02-03:
This is an intermitent error that appears to crash the `mavros_node`. More than once happened in the PX4 Jenkins CI (example: http://ci.px4.io:8080/blue/organizations/jenkins/Firmware/detail/PR-8804/1/pipeline/). 

A more deep search for the usage of `std::weak_ptr` leads me to think that the `std::bad_weak_ptr` exception is triggered because the `weak_link` for the `mavconn` `diagnostic_update` object that was deleted (based on the `std::bad_weak_ptr` exception definition, `std::bad_weak_ptr is the type of the object thrown as exceptions by the constructors of std::shared_ptr that take std::weak_ptr as the argument, when the std::weak_ptr refers to an already deleted object`). I'm pointing to a possible gap in https://github.com/mavlink/mavros/blob/master/mavros/include/mavros/mavlink_diag.h.

A possible solution will probably involve the exception catch and handling, besides adding a verification to `std::weak_ptr::expired` (http://en.cppreference.com/w/cpp/memory/weak_ptr/expired) in order to verify whether the managed object has already been deleted.

@dagar FYI.

vooon, 2018-02-05:
[std::weak_ptr::lock()](http://en.cppreference.com/w/cpp/memory/weak_ptr/lock) as you see already calls `::expire()`.
Problem maybe exists because of bug in stdcpp lib?

Message 
```
terminate called after throwing an instance of 'std::bad_weak_ptr'
what():  bad_weak_ptr
```

Says almost nothing.

TSC21, 2018-02-05:
Maybe run it with gdb would help?

vooon, 2018-02-05:
Maybe if you can reproduce.

TSC21, 2018-02-05:
This is almost unreproducible cause it is rare and intermitent under (what appears to be) the same test conditions. @dagar do we have a way of running a stress test for this using `gdb` on Jenkins?

1184, 2018-03-13:
@TSC21 hello,I have the same problem with you.How did you solve it?
I ues ros kinetic and gazebo7.When run "roslaunch rotors_gazebo fixed_wing_px4_sim.launch",the terminal will show me this messege:

terminate called after throwing an instance of 'std::bad_weak_ptr'
  what():  bad_weak_ptr
Aborted (core dumped)
[gazebo_gui-2] process has died [pid 20107, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/liaozhiyong/.ros/log/50d78bb0-2668-11e8-92f9-a0a8cda62f02/gazebo_gui-2.log].
log file: /home/liaozhiyong/.ros/log/50d78bb0-2668-11e8-92f9-a0a8cda62f02/gazebo_gui-2*.log

TSC21, 2018-03-13:
This is still an issue. What are the steps you take so to reproduce it? Does it happen often?

1184, 2018-03-13:
It happened every time.All I did like this:
1. $ mkdir ~/ws
2. $ cd ~/ws
3. copy the src.tar.gz and extract here
4. $ catkin_make
4. $ source devel/setup.bash
5. $ roslaunch rotors_gazebo fixed_wing_px4_sim.launch.

mhkabir, 2018-03-13:
@1184 Is that mavros crashing or Gazebo crashing?

1184, 2018-03-21:
@mhkabir I am gazebo7 and mavros( 0.23.3-0xenial-20180309-015020-0800),I am not sure if they crash.

TSC21, 2018-04-12:
@mhkabir can you give the steps to reproduce? I continue to point to a possible gap in https://github.com/mavlink/mavros/blob/master/mavros/include/mavros/mavlink_diag.h.

TSC21, 2018-04-12:
If you could do some startups with `gdb` on that may lead us to the problem source.

TSC21, 2018-05-10:
@mhkabir any updates on your side regarding this?

mhkabir, 2018-05-10:
Nope.

shylent, 2018-08-20:
At the moment I can consistently reproduce this problem at current master (a72ae7781fb78ab6c0f4dc829cf8943f1c5e8213) by running mavconn-test. I get

```
Error:   mavconn: tcp18: receive: End of fileterminate called after throwing an instance of '
         at line std::bad_weak_ptr'
219 in /home/mlvov/ros/ws/src/libmavconn/src/tcp.cpp
  what():  bad_weak_ptr
Aborted (core dumped)
```

shylent, 2018-08-20:
Further poking at it with gdb revealed the following:
```
(gdb) run
Starting program: /home/mlvov/ros/ws/devel/lib/libmavconn/mavconn-test --gtest_filter=TCP.client_reconnect
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Note: Google Test filter = TCP.client_reconnect
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from TCP
[ RUN      ] TCP.client_reconnect
[New Thread 0x7ffff6825700 (LWP 15147)]
[New Thread 0x7ffff5e12700 (LWP 15148)]
[New Thread 0x7ffff5611700 (LWP 15149)]
[Switching to Thread 0x7ffff5e12700 (LWP 15148)]

Thread 3 "mtcp1" hit Catchpoint 1 (exception thrown), 0x00007ffff73a18bd in __cxa_throw ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007ffff73a18bd in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007ffff7b5fab6 in std::__throw_bad_weak_ptr() () from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#2  0x00007ffff7b6bbc0 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count(std::__weak_count<(__gnu_cxx::_Lock_policy)2> const&) () from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#3  0x00007ffff7b8a8bb in std::__shared_ptr<mavconn::MAVConnTCPClient, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<mavconn::MAVConnTCPClient>(std::__weak_ptr<mavconn::MAVConnTCPClient, (__gnu_cxx::_Lock_policy)2> const&) () from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#4  0x00007ffff7b88765 in std::shared_ptr<mavconn::MAVConnTCPClient>::shared_ptr<mavconn::MAVConnTCPClient>(std::weak_ptr<mavconn::MAVConnTCPClient> const&) () from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#5  0x00007ffff7b86d55 in std::enable_shared_from_this<mavconn::MAVConnTCPClient>::shared_from_this() ()
   from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#6  0x00007ffff7b7f3ac in mavconn::MAVConnTCPClient::do_recv() ()
   from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#7  0x00007ffff7b8f117 in void std::_Mem_fn_base<void (mavconn::MAVConnTCPClient::*)(), true>::operator()<, void>(mavconn::MAVConnTCPClient*) const () from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#8  0x00007ffff7b8ebf9 in void std::_Bind<std::_Mem_fn<void (mavconn::MAVConnTCPClient::*)()> (mavconn::MAVConnTCPClient*)>::__call<void, , 0ul>(std::tuple<>&&, std::_Index_tuple<0ul>) ()
   from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#9  0x00007ffff7b8e4c3 in void std::_Bind<std::_Mem_fn<void (mavconn::MAVConnTCPClient::*)()> (mavconn::MAVConnTCPClient*)>::operator()<, void>() () from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#10 0x00007ffff7b8d80c in void boost::asio::asio_handler_invoke<std::_Bind<std::_Mem_fn<void (mavconn::MAVConnTCPClient::*)()> (mavconn::MAVConnTCPClient*)> >(std::_Bind<std::_Mem_fn<void (mavconn::MAVConnTCPClient::*)()> (mavconn::MAVConnTCPClient*)>&, ...) () from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#11 0x00007ffff7b8cb30 in void boost_asio_handler_invoke_helpers::invoke<std::_Bind<std::_Mem_fn<void (mavconn::MAVConnTCPClient::*)()> (mavconn::MAVConnTCPClient*)>, std::_Bind<std::_Mem_fn<void (mavconn::MAVConnTCPClient::*)()> (mavconn::MAVConnTCPClient*)> >(std::_Bind<std::_Mem_fn<void (mavconn::MAVConnTCPClient::*)()> (mavconn::MAVConnTCPClient*)>&, std::_Bind<std::_Mem_fn<void (mavconn::MAVConnTCPClient::*)()> (mavconn::MAVConnTCPClient*)>&) () from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#12 0x00007ffff7b8bbdf in boost::asio::detail::completion_handler<std::_Bind<std::_Mem_fn<void (mavconn::MAVConnTCPClient::*)()> (mavconn::MAVConnTCPClient*)> >::do_complete(boost::asio::detail::task_io_service*, boost::asio::detail::task_io_service_operation*, boost::system::error_code const&, unsigned long) ()
   from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#13 0x00007ffff7b60b80 in boost::asio::detail::task_io_service_operation::complete(boost::asio::detail::task_io_service&, boost::system::error_code const&, unsigned long) ()
   from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#14 0x00007ffff7b6370d in boost::asio::detail::task_io_service::do_run_one(boost::asio::detail::scoped_lock<boost::asio::detail::posix_mutex>&, boost::asio::detail::task_io_service_thread_info&, boost::system::error_code const&) () from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#15 0x00007ffff7b63131 in boost::asio::detail::task_io_service::run(boost::system::error_code&) ()
   from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#16 0x00007ffff7b639d4 in boost::asio::io_service::run() ()
   from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#17 0x00007ffff7b7df7a in mavconn::MAVConnTCPClient::MAVConnTCPClient(unsigned char, unsigned char, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned short)::{lambda()#1}::operator()() const () from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#18 0x00007ffff7b84cee in void std::_Bind_simple<mavconn::MAVConnTCPClient::MAVConnTCPClient(unsigned char, unsigned char, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned short)::{lambda()#1} ()>::_M_invoke<>(std::_Index_tuple<>) ()
   from /home/mlvov/ros/ws/devel/lib/libmavconn.so
#19 0x00007ffff7b84bb2 in std::_Bind_simple<mavconn::MAVConnTCPClient::MAVConnTCPClient(unsigned char, unsigned char, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned short)::{lambda()#1} ()>::operator()() () from /home/mlvov/ros/ws/devel/lib/libmavconn.so
[... cut ...]
#21 0x00007ffff73ccc80 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
---Type <return> to continue, or q <return> to quit---
#22 0x00007ffff78a16ba in start_thread (arg=0x7ffff5e12700) at pthread_create.c:333
#23 0x00007ffff6e3b41d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

Main point is that the exception is thrown from `shared_from_this`, which might happen either

* if object, from which it is called is _not_ managed by a shared_ptr, or
* if the weak_this inside enable_shared_from_this class instance somehow gets expired

theg4sh, 2018-12-07:
That issue could be closed due to #1078 in master already

closed 2018-12-07 by TSC21

asherikov, 2021-11-22:
The bug is still there, the reason must be playing with `this` in constructors 
https://github.com/mavlink/mavros/blob/ros2/libmavconn/src/tcp.cpp#L112
https://github.com/mavlink/mavros/blob/ros2/libmavconn/src/tcp.cpp#L334