mavlink/mavros #938: Ground Rover with Mavros

ChunJyeBehBeh, 2018-02-05:
### Issue details

Hey, I am a newbie to ROS but now I am learning use the ros and implement it on a rover and try to build a autonomous vehicle. 

I follow the tutorial below but I stuck in one of the steps already.
Tutorial Link: https://pixhawk.org/dev/ros/ground_rover#environment

I have git clone the mavros into my ros workspace already. May I know actuator_controls_publisher also inside the package already? May I know how to run it? Can I have more details about that?

Thanks for the help.

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic?
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ Y ] PX4

TSC21, 2018-02-17:
https://pixhawk.org/dev/ros/ground_rover is probably deprecated already. You should follow https://docs.px4.io/en/frames_rover/ together with https://dev.px4.io/en/ros/mavros_offboard.html. Though I am currently not sure what is the current support of the rover type for OFFBOARD control.

TSC21, 2018-05-04:
Closing as stale.

closed 2018-05-04 by TSC21