mavlink/mavros #941: Install an older release?

Changliu52, 2018-02-06:
Hi,
Just a quick question here. I don't use wstool or rosdep often.
Is there a systematical way to install an older version of the mavros?

I want to keep my development consistent by keeping my source unchanged. But now I am forced to update to the latest release every time when I reinstall my system following the installation instruction. 

Kind Regards,
Chang

vooon, 2018-02-07:
You may specify exact version to wstool (look at rosinstall file).

Changliu52, 2018-02-07:
Awesome. That work perfectly!
Thanks.

closed 2018-02-07 by Changliu52

mellonUAV, 2018-02-21:
Hello @TSC21,

Can you please elaborate on this, I am new to this stuff and have no idea what should I do to change from 0.22.0 to 0.18.3. I would really appreciate your help! 

Best regards,
Melih

TSC21, 2018-02-21:
https://github.com/mavlink/mavros/blob/master/mavros/README.md#source-installation

mellonUAV, 2018-02-21:
@TSC21,

In those instructions how do I specify version 0.18.3?

In /tmp/mavros.install file it says:

```
- git:
    local-name: mavlink
    uri: https://github.com/mavlink/mavlink-gbp-release.git
    version: release/kinetic/mavlink/2018.2.2-0
```
but nothing about mavros. I am really confused.